sipeed/MaixPy-v1 #165: len and spilt cannot work

crouse12, 2019-09-21:
The following examples cannot work.
a=repl,readline()
print(len(a))

a='Hello,World'
b=a.spilt(',')

robert-hh, 2019-09-21:
`a=repl,readline()`
indeed cannot work, because repl is not  a valid name, and readline() is not defined as such. readline() is defined a method of modules, like sys.stdin or of a file object, and can be usedm like in:
```
import sys
a = sys.stdin.readline()
```
Similar:
```
a='Hello,World'
b=a.spilt(',')
```
fails, because the method 'spilt' does not exist. You certainly wanted to use the split() method.
That all is standard Python behavior and not specific to MicroPython.

crouse12, 2019-09-21:
spilt exists in micropython for microbit, esp8266 and esp32.

https://techtutorialsx.com/2017/09/29/esp32-esp8266-micropython-string-split-method/

robert-hh, 2019-09-22:
You have a typo in the word: the method is call split(), not spilt(). And split() exists. Just look at the name of the link in your post.

closed 2020-08-07 by Neutree