sipeed/MaixPy-v1 #170: No keyboard input on serial connection to MaiXDUINO with MaixPy

jeancf, 2019-10-06:
I just received a new MAiXDUINO board. When I try to connect to it over serial cable I seen the welcome message below but the prompt does not register any of my keyboard inputs.

```
      [MaixPy] sd_init | SD_CMD0 is FF

 __  __              _____  __   __  _____   __     __
|  \/  |     /\     |_   _| \ \ / / |  __ \  \ \   / /
| \  / |    /  \      | |    \ V /  | |__) |  \ \_/ /
| |\/| |   / /\ \     | |     > <   |  ___/    \   /
| |  | |  / ____ \   _| |_   / . \  | |         | |
|_|  |_| /_/    \_\ |_____| /_/ \_\ |_|         |_|

Official Site : https://www.sipeed.com
Wiki          : https://maixpy.sipeed.com

MicroPython v0.3.2 on 2019-05-15; Sipeed_M1 with kendryte-k210
Type "help()" for more information.
>>>
```

I tried with 2 different linux machines, 2 different terminals: _screen_ (`sudo screen /dev/ttyUSB1 115200`) and _moserial_ and also on a windows 10 machine with _puTTY_. All exhibit the same behavior.

So I am stuck and it is very frustrating. Please help.

robert-hh, 2019-10-06:
With Windows Putty, did you disable flow control?
On linux you could try picocom. In my opinion this works better.

jeancf, 2019-10-06:
I tried picocom and minicom with the same result.

robert-hh, 2019-10-06:
I have no Maixduino here. Looking at the specs. I was wondering about the dual USB/UART bridge: Maybe it either provides two serial ports, or it has to be switched between the ESP32 and the K210.

jeancf, 2019-10-06:
There are indeed 2 serial ports: ttyUSB0 and ttyUSB1. What I describe above is on ttyUSB0. ttyUSB1 is silent. There are 4 LEDs on the board for TX/RX K210 and TX/RX ESP32. Only the LED RX K210 is on.

robert-hh, 2019-10-06:
When the K210 send data,  is the K210 TX LED flashing?. I would expect that this is the expected behavior: flash when data is transmitted, and otherwise off. A RX LED which is permanently would then indicate a faulty state. What happens if you supply the board though the DC in plug and pull of USB. Is the RX led then off?

jeancf, 2019-10-06:
The RX LED is solid RED all the time, even when powered through the DC plug.

[image]

robert-hh, 2019-10-06:
That has the faint smell of a hardware problem, in that the RX line is tied to GND. If you look at the board, are there any shorts of misplaced soldering particles in the region of the LED, the UART bridge? You may need a magnifying glass to tell.

jeancf, 2019-10-07:
I tried but the board is painted so the traces are barely visible.
Anyway, thanks for your help Robert.

robert-hh, 2019-10-07:
The problem could be at the SiPeed M1 module, that some solder went under the pad. Almost impossible to get it off. I would guess that the critical pin is at IO4 according to that pin map:
https://wiki.sipeed.com/en/maix/module/m1.html
You could check if at power off this pin is connected to GND. It should not.

jeancf, 2019-10-08:
My board is a [Maixduino](https://wiki.sipeed.com/en/maix/board/maixduino.html) so in the same logic, I tested if port 0 is connected to GND when powered off: I measure a 700 Ohms resistance.

robert-hh, 2019-10-08:
That's far too low. It should be open, in the MOhm range.

jeancf, 2019-10-08:
OK, so I guess this is a hardware problem most probably.

I wish somebody could confirm that when powered on and nothing connected, the RX K210 LED is off on a working board.

closed 2019-11-11 by jeancf