sipeed/MaixPy-v1 #171: kflash_gui error: attempting to use a port that is not open

jeancf, 2019-10-06:
To try and address [this issue](https://github.com/sipeed/MaixPy/issues/170) I wanted to know if upgrading the MaixPy firmware to v0.4 would fix the issue.

When I ran kflash_gui with elevated privileges

[image]

I got the following error:

[image]

So at this stage I really don't know what else to do to start using my MAiXDUiNO.

robert-hh, 2019-10-06:
Do you have any other application using /dev/ttyUSB0? When ouy say elevated privileges, does it mean you run as sudo?

jeancf, 2019-10-06:
I run it with sudo indeed and no other application is running.

ConnorChristie, 2019-10-11:
I had an issue where my board only accepted a Baudrate of 115,200. See if changing to that speed works

jeancf, 2019-10-11:
@ConnorChristie I tried several slower options with no luck.

Does your board have the RX K210 LED constantly on as soon as you power it up?

jeancf, 2019-11-11:
Board was defective.

closed 2019-11-11 by jeancf