sipeed/MaixPy-v1 #180: kpu_Out of memory

nnn112358, 2019-11-04:
The mobilenet 1000class example cannot be read after some version of maixpy.
Older versions of maixpy ​​are fine. I am using m5stickv.

・condition
>task = kpu.load("mbnet751.kmodel")
> or task = kpu.load(0x200000) 
>>SYSCALL: Out of memory
>ValueError: [MAIXPY]kpu: load error:6
https://github.com/sipeed/MaixPy_scripts/blob/master/machine_vision/mobilenetv1_1000class.py

・result:
>○maixpy_v0.4.0_44_g95f00f0_m5stickv.bin
○maixpy_v0.4.0_47_g39bb8bf_m5stickv.bin
×maixpy_v0.4.0_49_g8279a1f_m5stickv.bin
×maixpy_v0.4.0_82_gc3327b5_m5stickv.bin
http://dl.sipeed.com/MAIX/MaixPy/release/master/

Smeerak, 2019-11-04:
Does this happen even when you use the minimum binary? (maixpy_v0.3.2_minimum.bin)

nnn112358, 2019-11-05:
minimum.bin does not contain the necessary configuration for use with M5StickV.
Better to use with m5stickv.bin.

nnn112358, 2019-11-24:
Before maixpy_v0.4.0_47_g39bb8bf_m5stickv.bin is OK
Lastest version is Bad

### maixpy_v0.4.0_47_g39bb8bf_m5stickv.bin

>>kpu.memtest()
 ###free gc heap memory : 195 KB
 ###free sys heap memory: 3460 KB
>>task = kpu.load("mbnet751.kmodel")
 kpu_net __del__
>>kpu.memtest()
 ###free gc heap memory : 195 KB
 ###free sys heap memory: 680 KB

###  maixpy_v0.4.0_92_g639616a_m5stickv.bin
>>kpu.memtest()
 ###free gc heap memory : 36 KB
 ###free sys heap memory: 2580 KB
>>task = kpu.load("mbnet751.kmodel")
 E (3010759961) SYSCALL: Out of memory

nnn112358, 2020-01-03:
My problem is solved.
The problem was solved by adjusting the maixpy firmware configuration.

https://qiita.com/nnn112358/items/ef3435d113c03d44d574#m5stickv%E3%81%AE%E3%83%A1%E3%83%A2%E3%83%AA%E6%9C%80%E9%81%A9%E5%8C%96

closed 2020-01-03 by nnn112358