sipeed/MaixPy-v1 #182: kpu.load(0x300000) gives error: err: we only support V3 now, get V1263354956

hothanh, 2019-11-10:
Hello team,
We upload our kmodel network onto the Maix M1 Dock at address 0x300000. While running the kpu.load(0x300000) method we got the following result (network is trained with tflite framework and input image size is 160x160).
v=1263354956, flag=4, arch=0, layer len=1, mem=1926400, out cnt=127
err: we only support V3 now, get V1263354956
Do you have any idea about this error?
Thanks in advance

nnn112358, 2019-11-24:
NNCase v0.2.0 is generated kmodel V4.
NNCase Converter v0.1.0 RC5 is generated kmodel V3.

now maixpy is only kmodel V3.

What did your nncase version use?

hothanh, 2019-11-24:
Hello, thanks for your comment. We are using the newest release which is v0.2.0. This error did not happen on later version v0.1.0 rc5

nnn112358, 2020-01-03:
Maixpy only supports up to kmodel v3.

https://github.com/sipeed/MaixPy/issues/193

anfractuosity, 2020-06-09:
Just wondering if there are any plans to fix this issue?  I get it when i run any of the machine learning examples on the sipeed maixcube

Neutree, 2020-08-07:
try latest firmware, support v4 converted from nncase v2 bata2.0, if some question, open new issue~

closed 2020-08-07 by Neutree