sipeed/MaixPy-v1 #185: error loading cascade files

Feiko, 2019-11-14:
I've build a Haar Cascade and converted the cascade to a .cascade file using [cascade_convert.py](https://github.com/openmv/openmv/blob/master/ml/haarcascade/cascade_convert.py). When I try to load the cascade from sd or flash I get a core dump. I tried the default [haarcascade_frontalface_default.xml](https://github.com/openmv/openmv/blob/master/ml/haarcascade/cascades/haarcascade_frontalface_default.xml). I get the same problem. The default script where the haarcascade is loaded from memory does work `image.HaarCascade("frontalface", stages=25)` 
I can't find any prebuild .cascade file for testing. 

I'm using the following firmware version `MicroPython v0.4.0-82-gc3327b5 on 2019-10-25; Sipeed_M1 with kendryte-k210`

>>> import image
>>> image.HaarCascade('/sd/face.cascade')
core dump: misaligned load
Cause 0x0000000000000004, EPC 0x00000000800010c2
reg[00](zero ) = 0x0000000000000056, reg[01](ra   ) = 0x000000008000023e
reg[02](sp   ) = 0x000000008019c140, reg[03](gp   ) = 0x000000008018d390
reg[04](tp   ) = 0x0000000000000000, reg[05](t0   ) = 0x0000001000000000
reg[06](t1   ) = 0x00000000800010ac, reg[07](t2   ) = 0x000000000000007d
reg[08](s0/fp) = 0x000000008000f0de, reg[09](s1   ) = 0x0000000000000001
reg[10](a0   ) = 0x0000000000000004, reg[11](a1   ) = 0x000000008000f0de
reg[12](a2   ) = 0x000000008019c360, reg[13](a3   ) = 0x000000008019c460
reg[14](a4   ) = 0x0000000000000020, reg[15](a5   ) = 0x0000000000000003
reg[16](a6   ) = 0x0000000000000061, reg[17](a7   ) = 0x0000000000000000
reg[18](s2   ) = 0x000000008019c648, reg[19](s3   ) = 0x0000000000000000
reg[20](s4   ) = 0x00000000000000c0, reg[21](s5   ) = 0x00000000000000b0
reg[22](s6   ) = 0x00000000800e0f90, reg[23](s7   ) = 0x00000000802f3f1b
reg[24](s8   ) = 0x00000000800d3e98, reg[25](s9   ) = 0x00000000800d3e90
reg[26](s10  ) = 0x00000000800d3e88, reg[27](s11  ) = 0x00000000800d3e80
reg[28](t3   ) = 0x0000000000000020, reg[29](t4   ) = 0x0000000000000003
reg[30](t5   ) = 0x0000000000000004, reg[31](t6   ) = 0x0000000000000000
freg[00](ft0 ) = 0x0000000000000000(), freg[00]() = 0x00000000800d17b0()
freg[02](ft2 ) = 0x0000000000000000(), freg[00]() = 0x00000000800d17c0()
freg[04](ft4 ) = 0x0000000000000000(), freg[00]() = 0x00000000800d17d0()
freg[06](ft6 ) = 0x0000000000000000(), freg[00]() = 0x00000000800d17e0()
freg[08](fs0 ) = 0x0000000000000000(), freg[00]() = 0x00000000800d1808()
freg[10](fa0 ) = 0x0000000043028000(), freg[00]() = 0x00000000800d1838()
freg[12](fa2 ) = 0x0000000000000000(), freg[00]() = 0x00000000800d1860()
freg[14](fa4 ) = 0x0000000043020000(), freg[00]() = 0x00000000800d1870()
freg[16](fa6 ) = 0x0000000000000000(), freg[00]() = 0x00000000800d1880()
freg[18](fs2 ) = 0x0000000000000000(), freg[00]() = 0x00000000800d18a8()
freg[20](fs4 ) = 0x0000000000000000(), freg[00]() = 0x00000000800d18b8()
freg[22](fs6 ) = 0x0000000000000000(), freg[00]() = 0x00000000800d18c8()
freg[24](fs8 ) = 0x0000000000000000(), freg[00]() = 0x00000000800d18d8()
freg[26](fs10) = 0x0000000000000000(), freg[00]() = 0x00000000800d18e8()
freg[28](ft8 ) = 0x0000000000000000(), freg[00]() = 0x00000000800d1910()
freg[30](ft10) = 0x0000000000000000(), freg[00]() = 0x00000000800d1920()
W (21964026773) SYSCALL: sys_exit called by core 0 with 0x539`

Feiko, 2019-11-14:
I just tried loading the original [frontal.cascade](https://github.com/openmv/openmv/blob/master/scripts/unittest/data/frontalface.cascade) file from flash. `image.HaarCascade('/flash/frontalface.cascade')` throws the same core dump. So the issue is not the cascade file.

Neutree, 2020-08-07:
actually this func is not full tested, it's better to use model to find face https://github.com/sipeed/MaixPy_scripts/blob/master/machine_vision/demo_find_face.py

Feiko, 2020-08-08:
I agree, for finding faces using the KPU would make more sense. However my haarcascade can identify license plates and I don't have a suitable license plate database to train a new kmodel. Also I want to use the KPU to read the license plate. Hopefully this bug will be resolved soon, this project has been on hold for more than 6 months.

shanejohnpaul, 2020-09-22:
Any updates on this issue?

Feiko, 2020-10-19:
Any updates on when we can expect a fix? It would be very nice if we could use the Haarcascade.

cHemingway, 2020-12-13:
I am facing the same issue as well, on the M5StickV.

Firmware version `M5StickV_v5.1.2` Micropython prompt: `MicroPython v0.5.0-98-g7ec09ea22-dirty on 2020-07-21; Sipeed_M1 with kendryte-k210`