sipeed/MaixPy-v1 #187: Problem Recording Audio on py_video_record

Matias159753, 2019-11-29:
Im having problem when im trying to record the audio on the video module in maixpy (Maixpy-IDE).
Despite the fact that the video module has a "record" function that includes the option to record audio, if you set the bool variable to record to "true", it doesn't do anything.

Searching in the py_video.c source code, the C file that use the module for working, i figured out that the recording function only has "avi_record_append_video". This one only append the jpeg images that the camera captures, but it doesn't record the audio.

In the Header also exist the function "avi_record_append_audio", but when i try to use it, it doesn't work to me.

Can someone help me with this? I dont know how to modify or add this audio part to make it work. 😥

The following functions are the originals that Maixpy has on his Master Code.

**Video Recorder Function**
```
static mp_obj_t py_video_record(size_t n_args, const mp_obj_t *args, mp_map_t *kw_args)
{ 
    static mp_uint_t tim = 0;
    py_video_avi_obj_t* arg_avi = (py_video_avi_obj_t*)args[0];
    avi_t* avi = &arg_avi->obj;
    if( !avi->record )
        mp_raise_OSError(MP_EPERM);
    //TODO:resolve align error
    // if( !py_image_obj_is_image(args[1]) )
    //     mp_raise_ValueError("param must be image obj");
    image_t* img = (image_t*)py_image_cobj(args[1]); //TODO: optimize, add option get image dirrectly from sensor

    **int ret = avi_record_append_video(avi, img);**

    if( ret == 0)
        ret = -MP_EIO;
    if( ret <= 0 )
        mp_raise_OSError(-ret);
    //TODO: optimize
    while(mp_hal_ticks_us() - tim < avi->usec_per_frame);
    tim = mp_hal_ticks_us();
    return mp_obj_new_int(ret);
}
```

**avi_record_append_video**

```
int avi_record_append_video(avi_t* avi, image_t* img)
{
	int ret;
	uint8_t* buf = NULL;
	avi_data_t data = {
		.id = *((uint32_t*)AVI_VIDS_FLAG_TBL[0]),
		.len = 0,
		.data = buf
	};
	ret = video_hal_file_write(avi, (uint8_t*)&data, 8);
	if( ret <= 0 )
		return ret;
	ret = video_hal_image_encode_mjpeg(avi, img);
	if (ret < 0 )
		return ret;
	data.len = ret;
	ret = video_hal_file_seek(avi, -(data.len+4), VIDEO_HAL_FILE_SEEK_CUR);
	if( ret != 0)
		return -ret;
	ret = video_hal_file_write(avi, (uint8_t*)&data.len, 4);
	if( ret <= 0 )
		return ret;
	ret = video_hal_file_seek(avi, data.len, VIDEO_HAL_FILE_SEEK_CUR);
	if( ret != 0)
		return -ret;
	avi->content_size += 8 + data.len; //header length + data length
	++avi->total_frame;
	return data.len;
}
```
**avi_record_append_audio**

```
int avi_record_append_audio(avi_t* avi, uint8_t* buf, uint32_t len)
{
	avi_data_t data = {
		.id = (uint32_t)AVI_AUDS_FLAG_TBL[1],
		.len = len,
		.data = buf
	};
	int ret = video_hal_file_write(avi, (uint8_t*)&data, 8);
	if( ret <= 0 )
		return -ret;	
	ret = video_hal_file_write(avi, buf, len);
	if( ret <= 0 )
		return -ret;
	return 0;
}
```

Also another problem that I have is that i don't know how to get the information that the "avi_record_append_audio" needs. I mean, the only example that Maixpy has is a FFT example, that takes the audio from a I2S microphone.
I want to use this microphone too, but i don't think that the objects to use in that case.

-That is the i2s sound record function that the **FFT example** uses and it recongnize the audio.

```
STATIC mp_obj_t Maix_i2s_record(size_t n_args, const mp_obj_t *pos_args, mp_map_t *kw_args)//point_nums,time
{

    Maix_i2s_obj_t *self = pos_args[0];
    Maix_audio_obj_t *audio_obj = m_new_obj(Maix_audio_obj_t);
    audio_obj->audio.type = I2S_AUDIO;
    //parse parameter
    enum{ARG_points, 
         ARG_time,
    };
    static const mp_arg_t allowed_args[] = {
        { MP_QSTR_points, MP_ARG_INT, {.u_int = 0} },
        { MP_QSTR_time, MP_ARG_INT | MP_ARG_KW_ONLY, {.u_int = 0} },
    };
    mp_arg_val_t args[MP_ARRAY_SIZE(allowed_args)];
    mp_arg_parse_all(n_args-1, pos_args+1, kw_args, MP_ARRAY_SIZE(allowed_args), allowed_args, args);
    audio_obj->base.type = &Maix_audio_type;

    //compute buffer length
    if(args[ARG_points].u_int > 0)
    {
        if(audio_obj->audio.points > self->points_num)
        {
            mp_raise_ValueError("[MAIXPY]I2S:Too many points");
        }
        audio_obj->audio.points = args[ARG_points].u_int;
        audio_obj->audio.buf = self->buf;
    }
    else if(args[ARG_time].u_int > 0)
    {
        if(self->sample_rate <= 0)
            mp_raise_ValueError("[MAIXPY]I2S:please set sample rate");
        uint32_t record_sec = args[ARG_time].u_int;
        uint32_t smp_points = self->sample_rate * record_sec;
        if(smp_points > self->points_num)
            mp_raise_ValueError("[MAIXPY]I2S:sampling size is out of bounds");
        audio_obj->audio.points = smp_points;
        audio_obj->audio.buf = self->buf;
    }else 
    {
        mp_raise_ValueError("[MAIXPY]I2S:please input recording points or time");
    }
    //record 
    i2s_receive_data_dma(self->i2s_num, audio_obj->audio.buf, audio_obj->audio.points , DMAC_CHANNEL5);
    dmac_wait_idle(DMAC_CHANNEL5);//wait to finish recv
    return MP_OBJ_FROM_PTR(audio_obj);
}

```
Thanks you for your time and your help! =)
Matias.

yulonglovecode, 2020-04-28:
hi can i ask did you figure out how to record audio on maxipy IDE? please help

Neutree, 2020-08-07:
record audio in avi is not support yet, it's a new feature request, same as #246

closed 2020-08-07 by Neutree