sipeed/MaixPy-v1 #188: 使用MAiX GO双目摄像头无法复位

0x3A2B, 2019-11-30:
在运行`https://maixpy.sipeed.com/zh/libs/machine_vision/sensor.html`的例程2时
```
import sensor
import image
import lcd
import time
lcd.init()
sensor.binocular_reset()
sensor.shutdown(False)#选择sensor并初始化
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QVGA)
sensor.shutdown(True)#选择sensor并初始化
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QVGA)
sensor.run(1)
while True:
    sensor.shutdown(False) #选择sensor
    img=sensor.snapshot()
    lcd.display(img)
    time.sleep_ms(100)
    sensor.shutdown(True) #选择sensor
    img=sensor.snapshot()
    lcd.display(img)
    time.sleep_ms(100)
```
报错
`OSError: Reset Failed`
但是如果使用单镜头的代码运行时右边的摄像头可以正常工作，请问双目摄像头复位错误问题应该怎么解决？

Neutree, 2019-12-01:
检查摄像头硬件连接，然后试试单个摄像头，确认摄像头没坏

0x3A2B, 2019-12-04:
> 检查摄像头硬件连接，然后试试单个摄像头，确认摄像头没坏

两个摄像头都单独检查过，可以使用。双目摄像头和GO都是新买的，应该是好的。用万用表测PWDN电压确实会有变化，但是还是复位失败。

Neutree, 2019-12-04:
是不是gc0328,更新固件试试http://dl.sipeed.com/MAIX/MaixPy/release/master/

0x3A2B, 2019-12-05:
确实是固件问题，更新即可。

closed 2019-12-05 by 0x3A2B