sipeed/MaixPy-v1 #189: sensor.reset() OV2640 failed with error - Maxixduino

Razecus, 2019-12-02:
Hello I have the camera that was shipped with the maixduino, OV2640, and i get the error, from  the title, when i try to use the camera, after sensor.reset() i get the error. The camera was working before, and after some time stopped working. and i got the error. I had tried different ov2640, look like the camera ports on maixduino failed. (Since the camera works on other device).
 I had mesured the voltage on the camera regulators, and are ok, on the output of U3 3V0 and on the output of U12 1V3.

MengXiao92, 2019-12-31:
I met the same problem, can you solve it?

Razecus, 2019-12-31:
In my case, i believe it is ruined, i suspect that because camera was inserted with the board powered on, (my doter played with it ), and that caused the MCU I/O-ports to die in the M1 module. I have to confirm that, but i have no time at the moment, i plan to make a new pcb, for the M1 module, with protection on all the pins, so it can be plugged or unplugged while powered on. I will also like to have it more bulky, tinkerer friendly, with camera and display 2.54 mm pin headers, on board.

I do not know what happen in your case, with the board, may be camera was plugged while board was on,  or may be other stuff, but if you flash it with stock firmware, and it still acting like no sensor is present, than, this is a bad sign, look like is the same issue.

Neutree, 2020-08-07:
As @Razecus described, in most cases, it's hardware's issue, maybe connection not reliable may sensor broken.

Neutree, 2020-08-07:
I'll close this issue, if any problems, reopen it

closed 2020-08-07 by Neutree

valenzmanu, 2020-08-12:
I ran into the same issue, but I am sure the camera was not inserted while board (maixduino) was on.

It stopped working after I inserted an SD card (which is right next to the camera connector). The board was off when inserting the SD card. I tried with a new camera (OV7670 since I couldn't get a new OV2640) but got the error from the title. 

I'm guessing maybe some static electricity could have ruined the I/O pins (at the moment of inserting the sd card), but it's hard to tell.

Neutree, 2020-08-13:
@valenzmanu  OV7670 is not supported

valenzmanu, 2020-08-13:
@Neutree Thanks for the info. Then I'm not sure which of the two (camera or I/O pins) is not working.

marcusobrien, 2022-08-19:
Note that I had the same issue on both my Maixduino - its now solved

>>> import sensor
>>> import lcd
>>> import image
>>> lcd.init()
>>> sensor.reset()

If you see this response when doing a reset()

init i2c:2 freq:100000
[MAIXPY]: no sensor
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>

The you MAY have the pins of the camera the wrong way up. This Maixduino support OV2640 and the camera they ship now which is a ZV T01 GA4.4 - the copper pins need to be facing upwards away from the board.

[image]

justp7, 2023-05-10:
I also encountered this problem, by using a logic analyzer to analyze the camera pins is also no signal transmission, the official reply to me is that the IIC part of the chip is damaged and cannot be repaired.😢

magicianzz, 2023-08-29:
@marcusobrien Oh, my God. I didn't realize I had to wire it up

marcusobrien, 2023-12-28:
> @marcusobrien Oh, my God. I didn't realize I had to wire it up

My post doesn't recommend wiring it up it recommends the **correct** way to wire it up, your sarcasm is infintile, and mis-directed. You should be proud, you're the exact type of person that stops people adding to forums and publicly helping others. Must be great to be as good as you, not.