sipeed/MaixPy-v1 #191: sensor.set_windowing(roi) not working

bherbruck, 2019-12-09:
sensor.set_windowing does not center when no x, y coords are given,
also does not move window to x, y coords.
Same image is returned:
`sensor.set_windowing((50, 0, 224, 224))`
`sensor.set_windowing((0, 0, 224, 224))`

ted99tw, 2019-12-31:
I want this function too, but it seems the dvp lib leaves this as //to do in the source code. I use img.resize(224,224) instead.

bherbruck, 2019-12-31:
I haven't been able to get img.resize() working and can't find any documentation for it although it is in py_image.c.
I'm also using this for yolo2 and I think sensor.set_windowing also does something with dvp_set_ai_addr() which may affect yolo2. I have no clue though.

ted99tw, 2019-12-31:
It seems that yolo2 or mobilenet both refer to dvp_set_ai_addr() and the pix_ai data of the image itself.  What I am in need is to resize an image from QVGA to (224,224) for yolo2 or mobilenet by using resize(). Although resize() works fine but the ai data lost after resizing. So I make a #198 pull request to make this happen. I hope it can be accepted so. So we can just use image.resize(224,224) without using something like set_windowing(50,224,224) for kpu task.

Neutree, 2020-08-07:
gc0328 has supported, commit here : a3fc950007dffde5fccef65499181e65c18c1ff7
but other sensors need to implement

Neutree, 2020-08-07:
and pay attention, the width or height must multiple of 8 bytes