sipeed/MaixPy-v1 #193: kmodel v4 support?

kartben, 2019-12-16:
Hi,

Could you please share some pointer w.r.t how I could use a kmodel v4 (converted using nncase 0.2.x)?

Thanks!

nnn112358, 2020-01-03:
Maixpy ​​only supports up to kmodel v3.
MaixPY's kpu.load function is included in lib_sipeed_kpu.a, but the source has not been released.
I also expect kmodel v4 support.

windowsair, 2020-02-13:
Hey, here is a preview version:
https://bbs.sipeed.com/t/topic/1790
Unfortunately, it is not included in the latest build.

luismeruje, 2020-04-30:
Hello, I'm having the same problem. 
I tried compiling from source, since your latest commits talk about having added v4 support.
However, when I try to load my model I get a "core dump: misaligned load  ".
I also tried to compile my model with nncase 0.1.x but it says it does not support Conv2d layers.
Will I have to code in C?

Neutree, 2020-08-07:
try latest firmware, support v4 converted from nncase v2 bata2.0, if some question, open new issue~

closed 2020-08-07 by Neutree