sipeed/MaixPy-v1 #195: Try to increase speed of SDcard/TFcard

ZSYAD, 2019-12-25:
The speed of SDcard is too low, i want to record a video  to SDcard, the max frame rate is 12fps。

I find a strange code in sdcard.c :
static void sd_lowlevel_init(uint8_t spi_index)
{
gpiohs_set_drive_mode(SD_CS_PIN, GPIO_DM_OUTPUT);
spi_set_clk_rate(SD_SPI_DEVICE, 200000); /set clk rate/
}

Maybe the speed of SDCard are limited by this code.
Change 200000 to 10000000 may increase the speed
I want to change this code and rebuild to verify my guess, but i can not rebuild it. Could anyone help me to change this code and rebuild it?  Thanks very much.

Neutree, 2020-08-07:
SD的性能不一, 如果需要修改,可以按照 build.md 中所述对源码进行构建即可, 在 linux 或者macOS下编译,网络好的情况(最好有梯子)下很快就能构建成功

Neutree, 2020-08-07:
我先把issue关闭了,如果仍然存在问题,可以重新打开

closed 2020-08-07 by Neutree