sipeed/MaixPy-v1 #197: "SYSCALL: misaligned load recovered" and then "core dump: misaligned load"

eggfly, 2019-12-31:
V (4274273141) SYSCALL: misaligned load recovered at 8001e1a4. len:08,addr:0000002d,reg:06,data:0000000000000000,signed:1,float:0

core dump: misaligned load
Cause 0x0000000000000004, EPC 0x00000000800010c2
reg[00](zero ) = 0x0000000038000000, reg[01](ra   ) = 0x000000008000023e
reg[02](sp   ) = 0x000000008019ba60, reg[03](gp   ) = 0x000000008018d478
reg[04](tp   ) = 0x0000000000000000, reg[05](t0   ) = 0x0000000000000000
reg[06](t1   ) = 0x00000000800010ac, reg[07](t2   ) = 0x0000000000000001
reg[08](s0/fp) = 0x000000008001fcb6, reg[09](s1   ) = 0x0000000080224c20
reg[10](a0   ) = 0x0000000000000004, reg[11](a1   ) = 0x000000008001fcb6
reg[12](a2   ) = 0x000000008019bc80, reg[13](a3   ) = 0x000000008019bd80
reg[14](a4   ) = 0x0000000000000020, reg[15](a5   ) = 0x0000000000000003
reg[16](a6   ) = 0x0000000000000000, reg[17](a7   ) = 0x0000000000c65d40
reg[18](s2   ) = 0x000000008022114a, reg[19](s3   ) = 0x00000000800e0ba8
reg[20](s4   ) = 0x0000000000000047, reg[21](s5   ) = 0x00000000000000b0
reg[22](s6   ) = 0x00000000800e0b98, reg[23](s7   ) = 0x0000000080221100
reg[24](s8   ) = 0x0000000080218360, reg[25](s9   ) = 0x00000000800d3b80
reg[26](s10  ) = 0x00000000800d3b78, reg[27](s11  ) = 0x00000000800d3b70
reg[28](t3   ) = 0xfffffffffffffff1, reg[29](t4   ) = 0x0000000000000003
reg[30](t5   ) = 0x0000100000000000, reg[31](t6   ) = 0x0000008000000000
freg[00](ft0 ) = 0x0000000000000000(), freg[00]() = 0x00000000800d15d0()
freg[02](ft2 ) = 0x0000000000000000(), freg[00]() = 0x00000000800d15e0()
freg[04](ft4 ) = 0x0000000000000000(), freg[00]() = 0x00000000800d15f0()
freg[06](ft6 ) = 0x3461316500000000(), freg[-1073741824]() = 0x00000000800d1600()
freg[08](fs0 ) = 0x0000000000000000(), freg[00]() = 0x00000000800d1628()
freg[10](fa0 ) = 0x0000000042c80000(), freg[00]() = 0x00000000800d1658()
freg[12](fa2 ) = 0x000000003f800000(), freg[00]() = 0x00000000800d1680()
freg[14](fa4 ) = 0x0000000040a00000(), freg[00]() = 0x00000000800d1690()
freg[16](fa6 ) = 0x0000000000000000(), freg[00]() = 0x00000000800d16a0()
freg[18](fs2 ) = 0x0000000000000000(), freg[00]() = 0x00000000800d16c8()
freg[20](fs4 ) = 0x0000000000000000(), freg[00]() = 0x00000000800d16d8()
freg[22](fs6 ) = 0x0000000000000000(), freg[00]() = 0x00000000800d16e8()
freg[24](fs8 ) = 0x0000000000000000(), freg[00]() = 0x00000000800d16f8()
freg[26](fs10) = 0x0000000000000000(), freg[00]() = 0x00000000800d1708()
freg[28](ft8 ) = 0x0000000000000000(), freg[00]() = 0x00000000800d1730()
freg[30](ft10) = 0x0000000000000000(), freg[00]() = 0x00000000800d1740()
W (4367440110) SYSCALL: sys_exit called by core 0 with 0x539

我用MaixPy在M5StickV写了简单的app，用到了pgio.irq，然后按按钮次数比较多以后，就会在不稳定的时间出现这个问题，每次出现概率不同，现在心里比较崩溃，求高手看这个问题怎么分析，我随时配合～

eggfly, 2019-12-31:
看了下 不知道准不准：
1337是对应syscall.c的 sys_exit(1337);

recovered at 8001e1a4, 8001e1a4是在mp_obj_t mp_call_function_1(mp_obj_t fun, mp_obj_t arg);
EPC 0x00000000800010c2, 0x00000000800010c2也是在syscall.c里

eggfly, 2020-01-01:
把gpio.irq干掉好了。。

eggfly, 2020-01-01:
怀疑 这里或者相关代码，会造成内存污染

https://github.com/sipeed/MaixPy/blob/master/components/micropython/port/src/Maix/Maix_gpio.c

eggfly, 2020-01-01:
@Neutree 大佬求关注

Neutree, 2020-08-07:
这个问题一般是C层面触发了什么bug, 需要有可以复现问题的代码