sipeed/MaixPy-v1 #199: change model size

eraser3112, 2020-01-02:
The created ".kmodel" was tested with M5stickV.
[image]
error code is "model w=32, h=32".
Did the model conversion fail?
Conceivably can it be changed on maixpy?
Please tell me.

nnn112358, 2020-01-02:
Your maixpy image size is 320x240.
Your kmodel image size  is 32x32.
Both must be the same size.

eraser3112, 2020-01-02:
Thank you for your reply.
I don't want to change the size of the input image.
I want to change the size of the model.
The tool used for conversion is "nncase".
https://github.com/kendryte/nncase
The question location may be different.
Still, I think this is the only way to convert.
If you don't know, tell me your way.

nnn112358, 2020-01-03:
Can you post Kmodel source code with TensorFlow or Keras or caffe?
you need to change the first stage input size in source code.

eraser3112, 2020-01-04:
Here is what I used.
https://colab.research.google.com/drive/1Fqs5hqfk4J0j2Y0TeMfJTQaefelERbJn#forceEdit=true&sandboxMode=true&scrollTo=p8hOZWz19NcX
Rewrite the [.h5] file.
Others will cause errors in maixpy(ValueError).
After converting to tflite, use nncase.

AIWintermuteAI, 2020-03-14:
@eraser3112 While it's not completely impossible to change input size of the model(have a look at this article https://www.pyimagesearch.com/2019/06/24/change-input-shape-dimensions-for-fine-tuning-with-keras/ ), but the difference between networks trained on 224x224 and 32x32 would influence performance by a lot. You might be better off just fine-tuning MobileNet on a smaller dataset. You can have a look at my GitHub for examples.

Neutree, 2020-08-07:
there's 3 way to fix this error:
* change input size of image in maixpy by :
```
img = img.resize(32,32) # or copy function
img.pix_to_ai()
kpu.run_yolo2(...)
```
* change input size in pb file then convert to kmodel, as @AIWintermuteAI said
* change input size of your model in your train source code(model difinition)

closed 2020-08-07 by Neutree