sipeed/MaixPy-v1 #200: OSError: Reset Failed

rosariomoscato, 2020-01-03:
Hello everybody,
trying to run the "Hello World Example" on Maixpy Dock FW version **maixpy_v0.3.2_full** using MaixPy IDE 0.2.3, but always get OSError: Reset Failed.
Can anyone help? PLEASEEEEE

Thank you

TheImmigrant, 2020-01-11:
I had the same error , it was solved by re-connecting the camera.

rosariomoscato, 2020-01-12:
@TheImmigrant Thank you so much for yor reply. Actually I fixed the problem updating the firmware to the latest version.

bwellmeshed, 2020-01-13:
> @TheImmigrant Thank you so much for yor reply. Actually I fixed the problem updating the firmware to the latest version.

Which latest firmware did you use?

rosariomoscato, 2020-01-13:
maixpy_v0.5.0_12_g284ce83

Neutree, 2020-08-07:
most is firmware too old or hardware connection problem, or camera sensor broken, or firwamre not support your hardware, currently ov2640 gc0328 0v7740 is supported

closed 2020-08-07 by Neutree