sipeed/MaixPy-v1 #202: --toolchain option not working as described

smuehlst, 2020-01-04:
According to the instructions in `build.md` it should be possible to specify a custom path to the kendryte toolchain with `--toolchain`:

> Or you can customsize your toolchain path by
>
> python3 project.py --toolchain /opt/kendryte-toolchain/bin --toolchain-prefix riscv64-unknown-elf- config

This does not work as expected when I try this under Ubuntu. I have the toolchain installed in my HOME directory under `/home/stm/local/kendryte-toolchain`. Then I perform the following steps according to the instructions in `build.md`:

```
stm@COPERNICIUM:~/src/MaixPy/projects/maixpy_k210$ python3 project.py --toolchain /home/stm/local/kendryte-toolchain/bin
 config
-- SDK_PATH:/home/stm/src/MaixPy
maixpy
generate config file at: .config.mk
config complete
stm@COPERNICIUM:~/src/MaixPy/projects/maixpy_k210$ python3 project.py build
-- SDK_PATH:/home/stm/src/MaixPy
maixpy
build now
-- SDK_PATH:/home/stm/src/MaixPy
-- PROJECT_PATH:/home/stm/src/MaixPy/projects/maixpy_k210
-- Find component: /home/stm/src/MaixPy/components/boards
-- Find component Kconfig of boards
-- Find component: /home/stm/src/MaixPy/components/drivers
-- Find component Kconfig of drivers
-- Find component: /home/stm/src/MaixPy/components/kendryte_sdk
-- Find component Kconfig of kendryte_sdk
-- Find component: /home/stm/src/MaixPy/components/micropython
-- Find component Kconfig of micropython
-- Find component: /home/stm/src/MaixPy/components/spiffs
-- Find component Kconfig of spiffs
-- Find component: /home/stm/src/MaixPy/components/utils
-- Find component Kconfig of utils
-- find component: /home/stm/src/MaixPy/projects/maixpy_k210/main
-- Project defaults config file:/home/stm/src/MaixPy/projects/maixpy_k210/config_defaults.mk
-- python command: python3, version: Python 3.5.2

-- Load default: /home/stm/src/MaixPy/projects/maixpy_k210/config_defaults.mk
-- write  c header file  at: /home/stm/src/MaixPy/projects/maixpy_k210/build/config/global_config.h
-- Write makefile config at: /home/stm/src/MaixPy/projects/maixpy_k210/build/config/global_config.mk
-- Write  cmake  config  at: /home/stm/src/MaixPy/projects/maixpy_k210/build/config/global_config.cmake
warning: MIC_ARRAY_ENABLE (defined at /home/stm/src/MaixPy/components/kendryte_sdk/Kconfig:31) has leading or trailing whitespace in its prompt
CMake Error at /home/stm/src/MaixPy/tools/cmake/compile.cmake:244 (message):
  TOOLCHAIN_PATH set error:/opt/kendryte-toolchain/bin
Call Stack (most recent call first):
  CMakeLists.txt:29 (project)

-- Configuring incomplete, errors occurred!
See also "/home/stm/src/MaixPy/projects/maixpy_k210/build/CMakeFiles/CMakeOutput.log".
```
The specified toolchain path is in .config.mk:

```
stm@COPERNICIUM:~/src/MaixPy/projects/maixpy_k210$ cat .config.mk
# Generated by config.py, DO NOT edit!

CONFIG_TOOLCHAIN_PATH="/home/stm/local/kendryte-toolchain/bin"
```
But apparently it has no effect.

reynoldsbd, 2020-02-02:
Same issue here.

@Neutree looks like the issue was introduced with 5798d57. project.py is *always* passing `-DDEFAULT_CONFIG_FILE` to CMake, which causes compile.cmake to skip loading .config.mk

Neutree, 2020-08-07:
fixed in ef4ffc6300b0779d44bc70b06b40999101713d4b , set acoording to the build.md

closed 2020-08-07 by Neutree