sipeed/MaixPy-v1 #203: i2c does not work as intended

Dieter-Graef, 2020-01-10:
the reason is that in the file machine_i2c.c in line 302 and 303 the fpioa_set_function uses the variable self->i2c but this variable is allocated later in the program in line 307. Only I2C0 works therefor.
solution could be to change line 302 to:
fpioa_set_function(self->pin_scl, FUNC_I2C0_SCLK + i2c_id * 2);
and line 303 to:
fpioa_set_function(self->pin_sda, FUNC_I2C0_SDA + i2c_id * 2);

with best regards
Dieter Graef

bwellmeshed, 2020-01-13:
Would you happen to know where file machine_i2c.c is located at.

Dieter-Graef, 2020-01-13:
> Would you happen to know where file machine_i2c.c is located at.
here
\MaixPy\components\micropython\port\src\standard_lib\machine\machine_i2c.c
with best regards
Dieter Graef

bwellmeshed, 2020-01-13:
Okay thanks

Neutree, 2020-08-07:
fixed in commit 3f7f10269

closed 2020-08-07 by Neutree