sipeed/MaixPy-v1 #204: MLX90640 i2c support

thomaselemy, 2020-01-11:
Hi, i was wondering if support for the mlx90640 via i2c could be added. I can see there is support for the version of the board with the serial output but the sensor in its most basic form is i2c. It seems to be supported already on the openmv micropython port. I would be willing to learn how to add support for this if someone could assist me. Thanks

Neutree, 2020-01-15:
you can write driver based on micropython IIC driver `machine.I2C`, or C( [here](https://github.com/sipeed/MaixPy/tree/master/components/micropython/port/src/modules))

JuriShalkin, 2020-05-03:
Hello! I am also interested in connecting MLX90640 via I2C. Is there any success in this?

thomaselemy, 2020-05-06:
Hello. I never got around to it as I'm not sure where to begin. I hope there is more people who wish to see this feature available, I just dont have the skills to do it.

zzzeddy, 2020-05-30:
Hello, I also tried to use mlx90640 (adafruit Qwiic rev 2) on Maix Bit using i2c but failed (nothing from i2c device scan). I'm confused because it's working with AMG8833 (adafruit) using same i2c pin mapping. And I got temperature data using same device (mlx90640 sensor) with esp8266 or raspberry pi zero using i2c. Is there any solution for it?

thomaselemy, 2020-05-30:
@zzzeddy there needs to be a library written for the mlx90640, the timings and communication are different from the AGM8833, so unfortunately the mlx90640 is not working with the maix bit yet.

junhuanchen, 2020-11-13:
Now I2C add soft-ware i2c.

your can try this.

https://github.com/sipeed/MaixPy_scripts/blob/master/hardware/demo_i2c.py