sipeed/MaixPy-v1 #205: MaixPy UDP socket fails

Pali775, 2020-01-13:
with version of MaixPy ver:0.5.0_12, the UDP communication raises OSError: [Errno 5] EIO issue

example code:

//after NIC is initialized

```
host = “0.us.pool.ntp.org”
port = 123
addr = usocket.getaddrinfo(host,port)[0][-1]
NTP_QUERY = bytearray(48)
NTP_QUERY[0] = 0x1b
s = usocket.socket(usocket.AF_INET, usocket.SOCK_DGRAM)
s.settimeout(1)
res = s.sendto(NTP_QUERY, addr)
data, addr = s.recvfrom(1024)
s.close()
```

Pali775, 2020-03-21:
Trying with "MicroPython v0.5.0-31-gd3e71c0" the result is still the same...

> Connected: 192.168.1.41
> receive error: [Errno 5] EIO

Pali775, 2020-05-14:
Any advice?

Pali775, 2020-05-14:
Is it possible to implement this modification using https://www.freertos.org/FreeRTOS-Plus/FreeRTOS_Plus_UDP/FreeRTOS_Plus_UDP.html ?

Pali775, 2020-05-16:
I have just tried it with the last build, but the result is the same>
(MicroPython v0.5.0-69-g7928e0d on 2020-05-15; Sipeed_M1 with kendryte-k210)

Pali775, 2020-05-17:
Because still there is no solution (what a big mistake was to chose Sipeed Maixduino, anyway), I tried to debug the source.

As a result, it seems that the following function does not work properly:

```
int8_t esp32_spi_send_udp_data(uint8_t socket_num)
{
    esp32_spi_params_t *send = esp32_spi_params_alloc_1param(1, &socket_num);
    esp32_spi_params_t *resp = esp32_spi_send_command_get_response(SEND_UDP_DATA_CMD, send, NULL, 0, 0);
    send->del(send);

    if (resp == NULL)
    {
#if ESP32_SPI_DEBUG
        printk("Failed  get response\r\n");
#endif
        return -2;
    }

    uint8_t ok = resp->params[0]->param[0];

    if (ok != 1)
    {
#if ESP32_SPI_DEBUG
        printk("Failed to send udp data\r\n");
#endif
        resp->del(resp);
        return -1;
    }

    resp->del(resp);
    return 0;
}
```

The problem is that it returns with -1 so

`uint8_t ok = resp->params[0]->param[0];`

has problem. Accordingly, it seems that response is "wrong"...

Pali775, 2020-05-31:
Still analyzing the source, at this moment I localize the problem at the call of "**esp32_spi_send_udp_data**" function in "**STATIC mp_uint_t esp32_socket_sendto(mod_network_socket_obj_t *socket, const byte *buf, mp_uint_t len,  uint8_t* ip, mp_uint_t port, int *_errno)**" in file "**mod_esp32.c**"

"**esp32_spi_send_udp_data**" returns with -1 after calling "**esp32_spi_send_command_get_response**" with "**SEND_UDP_DATA_CMD**"

```
int8_t esp32_spi_send_udp_data(uint8_t socket_num)
{
    esp32_spi_params_t *send = esp32_spi_params_alloc_1param(1, &socket_num);
    esp32_spi_params_t *resp = esp32_spi_send_command_get_response(SEND_UDP_DATA_CMD, send, NULL, 0, 0);
    send->del(send);

    if (resp == NULL)
    {
#if ESP32_SPI_DEBUG
        printk("Failed  get response\r\n");
#endif
        return -2;
    }

    uint8_t ok = resp->params[0]->param[0];

    if (ok != 1)
    {
#if ESP32_SPI_DEBUG
        printk("Failed to send udp data\r\n");
#endif
        resp->del(resp);
        return -1;
    }

    resp->del(resp);
    return 0;
}
```

Any idea?

Neutree, 2020-08-07:
you should upgrade ESP32 firmware, maybe it can fix this error, firmware is here: https://github.com/sipeed/Maixduino_esp32_fimware/releases

I'll close this issue, if problem remain exists, reopen it so we can fix it

closed 2020-08-07 by Neutree

Pali775, 2020-08-31:
I have just tried it with the mentioned release and no change. UDP communication is still not available that is quite annoying.
Also tried with MicroPython v0.5.0-200-g8f6aa7b on 2020-08-31