sipeed/MaixPy-v1 #209: upip and ussl not supported

sasilva1998, 2020-01-16:
I need to install some libraries with upip, however it doesn't come included in maixpy, how can I get it inside?

I even tried just to put all upip scripts in so that I could maybe use it, however I need ussl as well, which doesn't come in it. Is there a way in which all this can be done? thanks for the help.

lypanov, 2020-02-29:
ussl issue is also referred to in https://github.com/sipeed/MaixPy/issues/105 

Unfortunately MaixPy doesn't seem to be a maintained product. An alternative could be https://github.com/loboris/MicroPython_K210_LoBo but I haven't yet managed to find the time to run the (poorly documented) install instructions. Regret purchasing my k210.

junhuanchen, 2020-07-01:
upip did not test.

ussl in https://github.com/sipeed/MaixPy/tree/develop.

you can compile it yourself.

[image]

valenzmanu, 2020-07-30:
Tried in develop branch but host name is not supported https://github.com/sipeed/MaixPy/issues/283

junhuanchen, 2020-11-13:
master support https.

maybe upip need test~

closed 2020-11-13 by junhuanchen