sipeed/MaixPy-v1 #217: binocular take snapshots form both sensors?

Vinschni, 2020-03-15:
Is there any support for taking images from both sensors in MaixPy?

junhuanchen, 2020-08-20:
There are two schemes in which the data path is double-channel and double-sensor rather than single-channel and double-sensor mode.

1. front and rear cameras.

```python

    try:
        sensor.reset(choice=1)
        sensor.set_pixformat(sensor.RGB565)
        sensor.set_framesize(sensor.QVGA)
        sensor.skip_frames(time=2000)
        for i in range(50):
            img = sensor.snapshot()
            lcd.display(img)
    except Exception as e:
        print(e)

    try:
        sensor.reset(choice=2)
        sensor.set_pixformat(sensor.RGB565)
        sensor.set_framesize(sensor.QVGA)
        sensor.skip_frames(time=2000)
        for i in range(50):
            img = sensor.snapshot()
            lcd.display(img)

    except Exception as e:
        print(e)
```

2. binocular camera

```python

import sensor
import image
import lcd
import time

lcd.init()

sensor.binocular_reset()
sensor.shutdown(0)  # 选中sensor 0
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QVGA)

sensor.shutdown(1)  # 选中sensor 1
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QVGA)
sensor.run(1)

while True:
    sensor.shutdown(0)  # 选中sensor 0
    img = sensor.snapshot()
    lcd.display(img)
    time.sleep_ms(100)

    sensor.shutdown(1)  # 选中sensor 1
    img = sensor.snapshot()
    lcd.display(img)
    time.sleep_ms(100)

```

All require hardware support.

junhuanchen, 2020-08-20:
If you have other problems, you can reopen it

closed 2020-08-20 by junhuanchen