sipeed/MaixPy-v1 #218: Maix Bit LCD won't show landscape display

zpowellman, 2020-03-17:
Hello All,
I have a Maxduino and a Maix Bit. Each is running the latest MaixPY firmware. I want to set the display to landscape mode. This code works great on the Maixduino but not on the Maix Bit.

import sensor, lcd
sensor.reset()
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QVGA)
#sensor.set_hmirror(1)
sensor.set_vflip(1)
sensor.run(1)
sensor.skip_frames()
lcd.init(freq=15000000)
lcd.direction(lcd.YX_LRUD) # 0x60
while(True):
lcd.display(sensor.snapshot())

I'm using the Maix PY IDE and the board types are set accordingly. I can't make the Maix bit go into landscape, regardless what I do. Some of the direction constants do weird things on both boards (i.e. 0x40, 0x80, 0xC0). Any ideas. I haven't looked into the code yet, maybe this is the time. I'm using the latest version of 5.0. Maybe a mistake?
Thanks,
Dan

krishnak, 2020-03-18:
I have this issue as well and I have the Maix dock, the LCD is always in portrait mode  when using camera feed. For images fed from files its in lanscape mode. I am coding in C and my lcd init and direction codes are like this

```
void lcd_init(void)
{
 
    uint8_t data = 0;
    tft_hard_init();
   /*soft reset*/
    tft_write_command(SOFTWARE_RESET);
    usleep(100000);
    /*exit sleep*/
    tft_write_command(SLEEP_OFF);
    usleep(100000);
    /*pixel format*/
    tft_write_command(PIXEL_FORMAT_SET);
    data = 0x55;
    tft_write_byte(&data, 1);
    lcd_set_direction(DIR_YX_RLDU);

    /*display on*/
    tft_write_command(DISPLAY_ON);
    lcd_polling_enable();
}

void lcd_set_direction(lcd_dir_t dir)
{
    lcd_ctl.dir = dir;
    if (dir & DIR_XY_MASK)
    {
        lcd_ctl.width = LCD_Y_MAX - 1;
        lcd_ctl.height = LCD_X_MAX - 1;
    }
    else
    {
        lcd_ctl.width = LCD_X_MAX - 1;
        lcd_ctl.height = LCD_Y_MAX - 1;
    }

    tft_write_command(MEMORY_ACCESS_CTL);
    tft_write_byte((uint8_t *)&dir, 1);
}

```

zpowellman, 2020-03-18:
Thanks for your response.  Does this code you supplied correct the situation?  I am assuming you are using the SDK, etc supplied by Sipeed?  Do they supply instructions for building from source?  What development environment are you using?
Thanks again,
Dan

krishnak, 2020-03-18:
I am not able to solve it, its in portrait mode both while using MaixPy and also while using the SDK, obviously some LCD driver code is incorrect. I hope some one answers here @Neutree

Neutree, 2020-03-30:
@zpowellman @krishnak  use `lcd.rotation(dir)` instead
dir: value from 0 to 3
```
img = image.Image(size=(80,40))
img.draw_rectangle(0,0, 80, 40, fill=True,color=lcd.RED)
lcd.rotation(1)
lcd.display(img)
```

zpowellman, 2020-03-30:
Thanks much.  Works nicely.
I appreciate your attention and response.

closed 2020-03-30 by zpowellman