sipeed/MaixPy-v1 #221: camera : Sensor time out error

nitz08, 2020-03-25:
In Arduino, I am getting "snap fail" in the serial monitor window. And black screen on LCD
For Maixpy IDE and Micropython, I am having the same issue.
If I modify the open-source python code (available on their website) to run only once (remove while loop), I am getting one picture on IDE and then error "Runtime Error: sensor timeout"

marcoA152, 2020-05-17:
same here
got "Runtime Error: sensor timeout" also with helloworld.py... tried many different firmwares but works only for few frames then stop... maybe my camera sensor is faulty? maybe something overheating?
any help is appreciated!

closed 2020-07-31 by Neutree

hktxt, 2021-03-28:
so, how to solve this? I encounter the same problem.

nitz08, 2021-03-28:
I changed the sensor and which worked for me.