sipeed/MaixPy-v1 #230: MaixPy IDE打不开（macOS系统）

luojia65, 2020-04-18:
错误提示：
```
Process:               MaixPy IDE [93616]
Path:                  /Applications/MaixPy IDE.app/Contents/MacOS/MaixPy IDE
Identifier:            org.openmv.openmvide
Version:               4.0 (4.0.2)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           MaixPy IDE [93616]
User ID:               501

Date/Time:             2020-04-18 22:56:42.167 +0800
OS Version:            Mac OS X 10.15.4 (19E266)
Report Version:        12
Anonymous UUID:        07301E68-4D00-6729-E601-8FB0EA3F5A35

Time Awake Since Boot: 540000 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x3] Wrong version

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: @rpath/QtWidgets.framework/Versions/5/QtWidgets
  Referenced from: /Applications/MaixPy IDE.app/Contents/MacOS/MaixPy IDE
  Reason: Incompatible library version: MaixPy IDE requires version 5.9.0 or later, but QtWidgets provides version 5.7.0

Binary Images:
       0x10fed9000 -        0x10feeaff7 +org.openmv.openmvide (4.0 - 4.0.2) <DB0EAD0D-A976-39FE-9AFF-B5C3CEEAD907> /Applications/MaixPy IDE.app/Contents/MacOS/MaixPy IDE
       0x10fef5000 -        0x10ff22ff3 +libExtensionSystem.1.0.0.dylib (0) <977EF7C9-BA61-31DE-9DAA-4315D1FCE2BC> /Applications/MaixPy IDE.app/Contents/Frameworks/libExtensionSystem.1.0.0.dylib
       0x10ff40000 -        0x10ff45ff7 +libAggregation.1.0.0.dylib (0) <3E579EE7-206A-3570-980F-CEAE4C8BC476> /Applications/MaixPy IDE.app/Contents/Frameworks/libAggregation.1.0.0.dylib
       0x10ff4f000 -        0x11004dff3 +libUtils.1.0.0.dylib (0) <0D5DA58D-2C73-3489-9338-A93D5A9F6550> /Applications/MaixPy IDE.app/Contents/Frameworks/libUtils.1.0.0.dylib
       0x112c42000 -        0x112cd3eff  dyld (750.5) <1F893B81-89A5-3502-8510-95B97B9F730D> /usr/lib/dyld
    0x7fff33cc1000 -     0x7fff34140ffb  com.apple.CoreFoundation (6.9 - 1675.129) <9E632A1E-9622-33D6-BCCE-23AC16DAA6B7> /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation

Model: iMac18,3, BootROM 181.0.0.0.0, 4 processors, Quad-Core Intel Core i5, 3.4 GHz, 8 GB, SMC 2.41f2
Graphics: kHW_AMDRadeonPro570Item, Radeon Pro 570, spdisplays_pcie_device, 4 GB
Memory Module: BANK 0/DIMM0, 4 GB, DDR4 SO-DIMM, 2400 MHz, 0x802C, 0x344154463531323634485A2D3247334232202020
Memory Module: BANK 1/DIMM0, 4 GB, DDR4 SO-DIMM, 2400 MHz, 0x802C, 0x344154463531323634485A2D3247334232202020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x16F), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1600.4)
Bluetooth: Version 7.0.4f6, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: APPLE HDD ST1000DM003, 1 TB
USB Device: USB 3.0 Bus
USB Device: Mi Dual Driver Earphones Type-C
USB Device: Bluetooth USB Host Controller
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB Debugger
Thunderbolt Bus: iMac, Apple Inc., 41.4
```

nongxiaoming, 2020-07-22:
我这边测试也是这个状态

Neutree, 2020-07-30:
下载新打包的版本: https://dl.sipeed.com/MAIX/MaixPy/ide/v0.2.5

closed 2020-07-30 by Neutree