sipeed/MaixPy-v1 #235: build 报错 fatal error: global_build_info_version.h: No such file or directory

4869z, 2020-04-29:
python3 project.py buil 后 报错
报错 提示 找不到 global_build_info_version.h
报错 位置 MaixPy/components/micropython/port/include/mpconfigport.h

compilation terminated.
/MaixPy/components/micropython/port/include/mpconfigport.h:4:10: fatal error: global_build_info_version.h: No such file or directory
 #include "global_build_info_version.h"

compilation terminated.
micropython/CMakeFiles/gen_qstr_i_last.dir/build.make:358: recipe for target 'micropython/CMakeFiles/gen_qstr_i_last' failed
make[2]: *** [micropython/CMakeFiles/gen_qstr_i_last] Error 1
CMakeFiles/Makefile2:619: recipe for target 'micropython/CMakeFiles/gen_qstr_i_last.dir/all' failed
make[1]: *** [micropython/CMakeFiles/gen_qstr_i_last.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

robert-hh, 2020-04-29:
Did you run
git submodule update --init 
After pulling the repository? That is typically missing in such cases.

4869z, 2020-04-29:
git submodule update --init 试过了 还是会报错
如果将 mpconfigport.h 中的 include两行注释
然后build 会报其他的错误
但是 在/projects/project_name/build/config 中会有一些生成的 .h文件
global_build_info_time.h
global_build_info_version.h
global_config.cmake
global_config.h
global_config.mk

是我执行顺序不对吗
我按着 官方给的 步骤来的：
cd MaixPy
cd projects/maixpy_k210
python3 project.py –toolchain /opt/kendryte-toolchain/bin –toolchain-prefix riscv64-unknown-elf- config
python3 project.py clean_conf
python3 project.py build

junhuanchen, 2020-07-01:
重来重建，这应该不是问题。

Neutree, 2020-07-31:
不用 clean_conf
直接 
```
git clone .... --recursive
cd MaixPy/projects/maixpy_k210
python3 project.py build
```

closed 2020-07-31 by Neutree