sipeed/MaixPy-v1 #237: OSError: Get version fail

wolfbomb, 2020-05-01:
> import network
> from Maix import GPIO
> from fpioa_manager import fm, board_info
> 
> WIFI_SSID = "*****************"
> WIFI_PASSWD = "*****************"
> 
> # IO map for ESP32 on Maixduino
> fm.register(25,fm.fpioa.GPIOHS10)#cs
> fm.register(8,fm.fpioa.GPIOHS11)#rst
> fm.register(9,fm.fpioa.GPIOHS12)#rdy
> fm.register(28,fm.fpioa.GPIOHS13)#mosi
> fm.register(26,fm.fpioa.GPIOHS14)#miso
> fm.register(27,fm.fpioa.GPIOHS15)#sclk
> 
> nic = network.ESP32_SPI(cs=fm.fpioa.GPIOHS10,rst=fm.fpioa.GPIOHS11,rdy=fm.fpioa.GPIOHS12, mosi=fm.fpioa.GPIOHS13,miso=fm.fpioa.GPIOHS14,sclk=fm.fpioa.GPIOHS15)
> 
> nic.connect(WIFI_SSID, WIFI_PASSWD)
> print(nic.ifconfig())
> print(nic.isconnected())
> print("ping baidu.com:", nic.ping("baidu.com"), "ms")
> nic.disconnect()
> 
But i am getting **OSError: Get version fail**

SSID and Password is definitely correct.

klebermagno, 2020-05-04:
same for me

Neutree, 2020-05-26:
update esp32 firmware please: https://github.com/sipeed/Maixduino_esp32_fimware/releases

Createyouraccount-alt, 2020-06-30:
Update does'nt help. Still same error. OSError: Get version fail
Updated to maixpy_v0.5.0_95_g05c4e70_minimum_with_ide_support.bin
OSError: Get version fail
Any help?

Createyouraccount-alt, 2020-07-10:
I was wrong. 
Correctly updating ESP32 firmware solved the problem.

closed 2020-08-06 by Neutree

ismasmenos, 2022-02-23:
It gives me the same error, with the same program, as the update did, since I have a t-Watch k210 which has similar characteristics to maixduino, but I do not know how to update the firmware of the esp32 without affecting the firmware of the k210 since they are independent right?