sipeed/MaixPy-v1 #241: demo script SpeechRecognizer hangs

IagniukovMaksim, 2020-05-07:
[image]

maixduino
MicroPython v0.5.0-62-gd6620f173

[MaixPy] i2s_num:0
<SpeechRecognizer>
Press the button to record the 1 keyword, the 1
...........---
modules_speech_recognizer_record

[MAIXPY]: record[0:0]
[MaixPy] speeking...

easy-and-simple, 2020-05-20:
As minimum button push handling code cant work in this way

easy-and-simple, 2020-05-25:
It crashes on record command
If record command is commented it runs cycle correctly 9 times but without any sense of course

shintarof, 2020-06-29:
I had a similar problem.
s.get_status() stays at 1 and doesn't change, so it's an infinite loop at line 95

lancewj, 2020-07-14:
你好请问有没有遇到：ImportError: cannot import name SpeechRecognizer这个错误

myuanz, 2020-07-15:
> 你好请问有没有遇到：ImportError: cannot import name SpeechRecognizer这个错误

默认固件没有开启这个功能, 现在需要自己编译固件.   
后面我们会在 lvgl 固件上默认开启这个功能.

nguyentaikhoa, 2020-09-17:
Hello,
I am also having the same problem. Is there anyone who can assist us?
Thạnk  you

myuanz, 2020-09-17:
> Hello,
> I am also having the same problem. Is there anyone who can assist us?
> Thank you

@Neutree

junhuanchen, 2020-12-11:
use https://github.com/sipeed/MaixPy_scripts/tree/master/multimedia/speech_recognizer

closed 2020-12-11 by junhuanchen