sipeed/MaixPy-v1 #242: 06e16a3 breaks kmodel v4 support

AIWintermuteAI, 2020-05-12:
Steps to reproduce the problem:
1) Download maixpy_v0.5.0_46_g06e16a3_minimum_with_ide_support.bin from https://cn.dl.sipeed.com/MAIX/MaixPy/release/master/maixpy_v0.5.0_46_g06e16a3 **OR compile it from source by doing git checkout 06e16a3 and then compiling with kmodelv4 support option enabled**
2) Upload Mobilenet(alpha=0.75) model (converted using nncase 0.2.0beta2) together with firmware
3) Run test script person_detector_v4fps.py
On loading model an error appears:
[image]
[person_detector_v4fps.zip](https://github.com/sipeed/MaixPy/files/4613163/person_detector_v4fps.zip)
[v4beta2_13.29397_FPS_74.9ms.zip](https://github.com/sipeed/MaixPy/files/4613169/v4beta2_13.29397_FPS_74.9ms.zip)

AIWintermuteAI, 2020-05-12:
https://github.com/AIWintermuteAI/aXeleRate/issues/5

corleonechensiyu, 2020-05-18:
@AIWintermuteAI i have the same error when i use the maixpy_v0.5.0_46* even the latest version .But version maixpy_v0.5.0_31_gd3e71c0 can pass.

AIWintermuteAI, 2020-05-18:
Yes, correct. I know versions before v0.5.0_46 can be used normally :) I just want to let sipeed team know that v0.5.0_46 causes issue, that they need to solve.

AIWintermuteAI, 2020-05-26:
Kmodel v4 (converted with nncase beta2) can be used normally in https://github.com/sipeed/MaixPy/commit/461f841fe95cf9109df32e1f3a76d60a659c31af 
Issue closed, good job @Neutree

closed 2020-05-26 by AIWintermuteAI

anhnguyen2345, 2020-07-28:
How can I do "Upload Mobilenet(alpha=0.75) model (converted using nncase 0.2.0beta2) together with firmware" in https://www.instructables.com/id/Object-Detection-With-Sipeed-MaiX-BoardsKendryte-K/. thanks

AIWintermuteAI, 2020-07-30:
Use https://github.com/sipeed/kflash_gui tool, choose firmware file(flash it to 0x00000) and model file(flash to 0x200000 for example), then flash.