sipeed/MaixPy-v1 #243: SXGA ENOMEM error

KBacsa, 2020-05-12:
While attempting to run the following micropython code on the sipeed MAix Bit : 
```
import sensor
import lcd
import time

clock = time.clock()
lcd.init()
sensor.reset()
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.SXGA)

while True:
    clock.tick()
    img = sensor.snapshot()
    lcd.display(img)  
```

I get the following error :
``` 
init i2c2
[MAIXPY]: find ov2640
True
␛[0;31mE (5033433707) SYSCALL: Out of memory
␛[0m
Traceback (most recent call last):
  File "<stdin>", line 9, in <module>
OSError: [Errno 12] ENOMEM

```

There seems to be a problem regarding the support of the SXGA resolution.

KBacsa, 2020-05-12:
The same error can be reproduced when attempting to use the UXGA format i.e. : 
```
sensor.set_framesize(sensor.UXGA)
```

luismeruje, 2020-05-13:
I've had the same problem. As of now, they don't support resolutions higher than VGA due to memory limitations, even though the camera sensor is capable of higher resolutions.

Neutree, 2020-08-06:
QVGA is recommend, SXGA is not implement, it can be a new feature