sipeed/MaixPy-v1 #245: can't display preview video in IDE

neolu, 2020-05-15:
can't display preview video in IDE

galaxystar1yuji, 2020-06-04:
When I connect Maix Bit to USB2.0 port, I cannot preview video in MaixpyIDE.
But when I connect to USB3.0 port, I can preview.
This phenomenon occur both windows and linux. I don't know the reason.

Neutree, 2020-08-06:
Maybe data lose, parity fail, so not display image on IDE, but we failed to reproduce this problem

Neutree, 2020-08-06:
Maybe as @galaxystar1yuji saied, replace a reliable USB cable and change port or PC will resolve this problem, we will continue pay attention to this problem

zaferAt, 2020-11-27:
Same here with my Maixduino when it is connected to my 1st gen AMD Ryzen machine. But in my case usb 2 ,3 or 3.1 ports didn't change anything. On the other hand, when I connected to my old 2. gen Intel laptop, everything works fine, every time. 

**Workaround**: connect to your device as usual by selecting the lower numbered COM port (e.g select "com 1" if there is "com1" and "com 2") and start the device by pressing play button. Once device started go to Tools -> Open Terminal -> New Terminal.  "Connect to serial port" -> "Please select a serial port" -> "com 2" (the remaining com port) -> "please enter the baud rate" -> "115200".

Preview video will start working as it was supposed to be. A second window will open. That window needs to stay open.

remixer-dec, 2021-03-24:
Similar issue here. Latest firmware (v0.6.2_41_g02d12688e), IDE v0.2.5. Using USB2.0 I can see the image stream up to QVGA-RGB, however only in a dark room, adding more light causes the stream to freeze / not receive any new frames. VGA doesn't work at all. With USB3.0 for some reason I'm only able to get QQVGA-GRAYSCALE, everything above that doesn't show up. With USB3.1 a COM port doesn't even appear and after a few seconds windows says that an unknown device has been detected. Device manager also shows that "the request to set the USB address failed".  
To check that this problem is not on my side, I tried two different cables, and using the same cables, I'm able to use the same 3.0/3.1 ports with another board (m5stickC).  
Also I don't really understand the workaround from previous comment, since I only have 1 com port available for 1 device.

romainreignier, 2021-09-07:
I have the same issue on my AMD Ryzen 2400G computer, I had read somewhere that there is an issue with the OpenMV IDE and AMD cpus but cannot find the page anymore.

By the way, where is the MaixPy IDE source code so I ca try to debug it?

niallofitz, 2024-01-03:
> can't display preview video in IDE

i think there might be a simple fix for this. I was having same issue and noticed a small left pointing arrow on the right hand side of the maixpy ide screen. the video preview and colour breakdown is hidden in there. it seems to rehide this every time the program is minimised too!! way too much time wasted on this!!