sipeed/MaixPy-v1 #246: AVI video recording without sound and can't be played

easy-and-simple, 2020-05-16:
Hello,
AVI video is recorded without sound and video recorded using video recording demo script have no index and can't be played
I tried to assign MIC0 pins to I2S device 0 channel 0 but this did nothing
Also documentation is awful ant there is no any example how to use microphone. Speach recognition is not working of course because firmware does not contain such module as SpeechRecognizer

anoken, 2020-05-24:
Unable to play  in demo_video_play.py the file recorded by demo_video_record.py.

>>demo_video_play.py
>>OS Error :6

demo_video_record.py
https://github.com/sipeed/MaixPy_scripts/blob/master/multimedia/video/demo_video_record.py

demo_video_play.py
https://github.com/sipeed/MaixPy_scripts/blob/master/multimedia/video/demo_video_play.py

Lastest MaixPy (MicroPython v0.5.0-78-gfc3aae3eb) is a bug.
No header information in demo_video_record module.

This cause is 2d307ae47bc8c274cd6cb4c55e26c090abdfe6ba.
There was no problem before this commit.
https://github.com/sipeed/MaixPy/commit/2d307ae47bc8c274cd6cb4c55e26c090abdfe6ba

It is better to revert to the previous change.

junhuanchen, 2020-05-26:
> Unable to play in demo_video_play.py the file recorded by demo_video_record.py.
> 
> > > demo_video_play.py
> > > OS Error :6
> 
> demo_video_record.py
> https://github.com/sipeed/MaixPy_scripts/blob/master/multimedia/video/demo_video_record.py
> 
> demo_video_play.py
> https://github.com/sipeed/MaixPy_scripts/blob/master/multimedia/video/demo_video_play.py
> 
> Lastest MaixPy is a bug.
> No header information in demo_video_record module.
> 
> This cause is [2d307ae](https://github.com/sipeed/MaixPy/commit/2d307ae47bc8c274cd6cb4c55e26c090abdfe6ba).
> There was no problem before this commit.
> [2d307ae](https://github.com/sipeed/MaixPy/commit/2d307ae47bc8c274cd6cb4c55e26c090abdfe6ba)
> 
> It is better to revert to the previous change.

Thank you very much for your question.

This [2d307ae](https://github.com/sipeed/MaixPy/commit/2d307ae47bc8c274cd6cb4c55e26c090abdfe6ba) fix is a change to make the normal player play normally. If you need to restore the previous function, it will be configured to allow recording.

Avi -> record_audio = args_parsed [ARG_audio] u_bool;

So if you want to be able to be played by record to play as before, you need to set it

V = video.open ("/sd/capture.avi", record=1, record_audio = true, interval=200000, quality=50)

There is a bug here, because there is no audio header, so it cannot be played by normal players.

To solve this problem, we should further fix the case where the play parse does not have audio headers.

junhuanchen, 2020-05-26:
> Hello,
> AVI video is recorded without sound and video recorded using video recording demo script have no index and can't be played
> I tried to assign MIC0 pins to I2S device 0 channel 0 but this did [...]

1. Speechrecognizer is being tested internally. It does not incorporate implementation and provide use cases. Where did you know about this module.

2. AVI does not realize audio recording function, so as you said, avi video recording without sound and can't be played, but it can already be played (normal video player, but no sound).

AndreiUngureanu101, 2020-05-27:
Can I play a recorded video and treat it as a camera capture in order to run machine learning models on it?

easy-and-simple, 2020-05-27:
anoken
Thank you for info about header issue.
I will not start to fix it because I have no idea what they are doing and why, but this doesn't make your replay less useful.

junhuanchen
speachrecognizer is available as not working example in maixpy demo scripts that are available for download here on github
If avi without sound is dumb but acceptable case, lack of microphone example is not acceptable at all. Why boards have microphone if they can't be used. I hope buggy speachrecognizer example is not burned my mic because it connects mic data line to I2S output data line instead to input data line.
Also there is no option to append avi video that is very necessary option

AndreiUngureanu101
To analyze video it is necessary to have function that return single frame, but it look that there is no such function, so you can't analyze avi on maixpy boards. Also I dont understand from where you will get avi with resolution 320x240, and what is the sense to analyze it on maixpy if you can do that on pc or phone

AndreiUngureanu101, 2020-05-27:
@easy-and-simple Well the idea was to run everything on the micropython environment on the device without compiling anything for the pc. You could do workarounds but this feature would be helpful.

junhuanchen, 2020-05-27:
> anoken
> Thank you for info about header issue.
> I will not start to fix it because I have no idea what they are doing and why, but this doesn't make your replay less useful.
>
> junhuanchen [...]

No, I mean you can put the recorded avi on a PC to determine the problem, and make the sample public so that others can help you analyze and solve the problem.

Speachrecognizer has not yet merged this code in the firmware, and I think there is a misunderstanding.

Your focus should be on solving problems, not finding meaning.

AndreiUngureanu101, 2020-05-27:
@junhuanchen what about loading the video frame by frame? is it possible?

junhuanchen, 2020-05-28:
> @junhuanchen what about loading the video frame by frame? is it possible?

That's true.

see this https://stackoverflow.com/questions/885160/what-is-the-specifications-for-motion-jpeg

Neutree, 2020-08-06:
* bug: can not play record avi video, solved
* bug: some player can not play video recorded by maixpy, resolved, for no audio data but audio header exist, so just remove audio header can resolved this problem, or add audio data, but it should be a new feature, so this bug resolved
* new feature request: add record audio when record avi video, not implemented
* new feature: add API for video module, `video.get_frame()` to get one frame as a image object for some usage, not implemented