sipeed/MaixPy-v1 #247: What mean digits in yolo anchor set that are used in object detection examples?

easy-and-simple, 2020-05-17:
Hello Guys,
Does someone know what mean digits in yolo anchor set that is used in object detection examples?
For example digits in face detection anchor set of 10 digits:
anchor = (1.889, 2.5245, 2.9465, 3.94056, 3.99987, 5.3658, 5.155437,
          6.92275, 6.718375, 9.01025)  # anchor for face detect

AIWintermuteAI, 2020-05-18:
These are anchors, a preset "boxes" for YOLO v2 object detection model. The answers about their existence and meaning can be found at original Darknet YOLO repository or on google, if you search for "yolo anchors explained"
https://github.com/pjreddie/darknet/issues/568

easy-and-simple, 2020-05-19:
I already know what anchors are, I want to know what these 10 digits mean
Thread you gice as some type of example is really useless because there no any explanation of anything, just dumb empty talks.
I am asking here so dont send me to search in this dumb and useless google, because I don't want to waste years looking for useful info like you did. I am not surprised that you still have not learned what they mean after years searching in garbage google supply

easy-and-simple, 2020-05-20:
Just FYI, I want to inform you that I already know what these digits mean and how to obtain?calculate their values, so dont think if you not tell I will not learn. Problem is that I wasted few minutes, but I waste these minutes each time I need to learn new things so in summary I will waste maybe about month. Month could not look much time for you, but I am interested in many things and cant waste month for nothing on each of them

Vinschni, 2020-05-22:
Dear easy-and-simple. The anchors are specific to your model.
An introduction can be found here: 
https://medium.com/@andersasac/anchor-boxes-the-key-to-quality-object-detection-ddf9d612d4f9

Issue can be closed.

easy-and-simple, 2020-05-22:
I know this article, but there are explained anchor boxes and I need to know about yolo 2 digits.
I also know that these 10 digits are 5 couples of weight and high of anchor boxes, but I still have no idea in what units are declared these dimensions. in yolo2.
For yolo3 I know that dimensions are in pixels, but for yolo2 cant find info about units

Neutree, 2020-08-06:
pixels / 32

Neutree, 2020-08-06:
e.g.

anchors:
[5.25, 5.218749999999999, 1.78125, 1.78125, 2.4999999999999996, 2.4999999999999996, 4.21875, 4.21875, 3.3125, 3.28125]

Boxes:
 w: [168.  57.  80. 135. 106.]
 h: [167.  57.  80. 135. 105.]
 (w, h): [168. 167.  57. 5.  80. 80.  135. 135. 106. 105.]

closed 2020-08-06 by Neutree