sipeed/MaixPy-v1 #248: Sipeed INTENTIONALY blocks KPU and machine vision feature of MAIX boards!!!

easy-and-simple, 2020-05-17:
Sipeed,
Don't be so greedy!!!!
We buy these boards for not small money so don't block KPU and machine vision!
KPU and machine vision are working only with encrypted models that need board key
If I try to use my model it is not working because model cant be loaded and board just hangs on model load and disconnects from ide
This is really insolent because your machine vision demos are just ports to micropython of Kentryte examples
Remove your micropython restriction for using unencrypted models now!!!

AIWintermuteAI, 2020-05-18:
While they do encrypt some of their models, I think the issue you're talking is a bug I described in https://github.com/sipeed/MaixPy/issues/242 . Hopefully it will get fixed soon. Until then I am using firmware prior v0.5.0_46.

easy-and-simple, 2020-05-18:
In your case at least you get some error.
In my case it just freezes
I have installed latest build, but have no any idea what version of kmodel is used because there is no any description what is new in these builds. Blind pick latest with obviously wrong hope it is best
I am trying with yolo20 model from Kendryte examples. Inside examples have supplied kmodel, I also converted supplied tflite model with nncase but it always freeze in same way
Obviously their model encryption is root of this issue
We pay for these boards cost as for Intel® B460 Chipset, so blocking KPU is really greedy and insolent, because on top of all we are wasting our time.
Thank you for advise to use firmware version before 0.5.0_46. I will try to see what will happen

AIWintermuteAI, 2020-05-18:
You can try connecting to the board using serial monitor and executing your script from there - ionly then you can see error message like in my screenshot. 
yolo20 model from Kendryte examples is kmodel v3. What version of nncase do you use?

easy-and-simple, 2020-05-18:
Latest of course hahah, 
but for windows= Should be 0.2 beta 4
Why you think its matter? In example is supplied and kmodel, same thing happen
I open console that is in ide before to run script it should be the same as it is executed from console
Now with 0.5.0_42 I get error no enough memory

easy-and-simple, 2020-05-18:
With 5.0_42 minimum model is loaded but then on code = kpu.run_yolo2(task, img) issue OSError: run error: 2013
Do you have idea where is problem now?

AIWintermuteAI, 2020-05-18:
Erm, I haven't tried beta4 and using beta2 currently. If you're using latest beta versions, be prepared for bugs :)
Executing your script in serial monitor and with "Run Script" is slightly different for debugging purposes - if you use "Run Script" you won't get traceback from the errors C code is throwing, only Micropython errors.
Consult this issue https://github.com/AIWintermuteAI/aXeleRate/issues/8
and have a look at the examples here:
https://github.com/AIWintermuteAI/aXeleRate/tree/master/example_scripts/k210

run error: 2013 is caused by not setting outputs for the network
https://github.com/AIWintermuteAI/aXeleRate/blob/bb7f3deda509d7f310b231fa9a39d0b529eb35f5/example_scripts/k210/detector/racoon_detector.py#L14

xddcore, 2020-05-18:
hi,bro.
I encountered the same problem as you last month. When loading the yolo model, an error occurs or the serial port fails. **First of all, it is worth mentioning that the generation of this problem has nothing to do with the model encryption adopted by sipeed**.You can try to solve this problem from the following aspects:
> 1. The network output feature map is HWC
> 2. Replace the MIX version of Maixpy firmware to load the model (the serial port fails, largely due to 2MB kpu memory overflow)
> 3. Make sure that the input image size is consistent with the model input layer size.

easy-and-simple, 2020-05-18:
Guys,
Firmware download from sipeed side is not working anymore

AIWintermuteAI how I can execute script in serial monitor?
In documentation for KPU have no such function as set_output
How I could know it exists and how to use it?
What mean 0, 7,7,30 in your example? According to xddcore it should be H, W, 5 * (4 + 1 + class_number) thats are 3 parameters and you have 4 parameters?

xddcore why you bolded that problem is not in sipeed encryption if their models are working, but other models are not working? Of course problem is in their encryption. Maybe problem with blocked firmware download is also not in sipeed?!
Model in Kendryte yolo example and in sipeed yolo demo is the same yolo20.kmodel but sipeed version is 1KB smaller and working while kendryte model is not working with sipeed maixpy firmware
Also I am new to AI and don;t know what means NMS and so on

Also I have no idea how to know model structure if I have not build it and it is supplied prepared for use with k210
Documentation supplied by sipeed id awful - bad english, obviously auto translated from chinese, parameters are not explained, no examples= There is 0 examples how to use build-in microphone. All is very bad

luismeruje, 2020-05-18:
@easy-and-simple I don't think they are "encrypting" their models.
They are just really REALLY bad at documentation and general support.
I've managed to train my own mobilenet classifier model, however it was not easy.
I followed this tutorial:
https://blog.sipeed.com/p/680.html
Plus, I had to find the correct version of the firmware. The only one that worked for me was the one provided as a zip file in this thread:
https://en.bbs.sipeed.com/t/topic/1790

Finally, the model only works when you use inference-type=uint8 on ncc else the model will be too big to load, and the only version of ncc that worked for me was version 0.2.0-beta2, just like @AIWintermuteAI mentioned.

I really hope they start at least answering some of the open issues here on github and start doing firmware releases through github again.

easy-and-simple, 2020-05-18:
Oh no, this is real nightmare
I checked links luismeruje supplied
nncase 0.2.0-beta2 is same garbage as beta4 because from version 0.2 is produced kmodel v4 that is 4 time slower than v3. WTF?! What means "Support more ops" if it is 4 times slower?! Support on CPU instead on KPU?! It is hopeless. In China obviously still don't know that new versions must improve and not destroy performance. And where is benefit from KPU if data is processed by CPU?! What is sense to buy K210 based boards if KPU is just fancy word?! Why these boards cost so much if they are nor better that ESP32 or STM32?
Initially they made buggy processor K210 and then insted to compensate it with software they make even more buggy software without documentation and even info file telling us what is new in these hundred builds.
Guys, if for each model they supply different firmware, that means encrypted models.
So we need to unite and demand unrestricted firmware for non-encrypted models!
Lets they supply special firmware only for their encrypted models

xddcore, 2020-05-19:
> Guys,
> Firmware download from sipeed side is not working anymore
> AIWintermuteAI how I can execute script in serial monitor?
> In documentation for KPU have no such function as set_output
> How I could [...]

hi.
>1. Regarding the serial port monitoring software, you can use putty to monitor the serial port output of the development board type-c port.
>2. H, W, 5 * (4 + 1 + class_number), this is the definition specification of yolov2 network output feature map. Where H represents the height of the feature map and W represents the width of the feature map. 5 represents five anchors, 4 represents cx, cyw, h. 1 represents confidence, and class_number represents the number of categories.
> 3. As far as I understand, * .kmodel is not affected by the encryption firmware strategy, only * .smodel will be affected by this strategy.
> 4.NMS (Non Max Supression), the details and functions of this algorithm can be consulted on google.
> 5. If the problem still exists, you can send the problematic kmodel to the email 1034029664@qq.com. I will try to help you reproduce the problem.

Good Luck!

easy-and-simple, 2020-05-19:
hi @xddcore,
As I said I am new and this means new to all of AI(less than 1 week), so I need some working base from where to start.
If you have enough knowledge could you help me with some basics as:
1--- recording sound from build in microphone and recording video with sound
2--- understanding yolo2 features of KPU as what means these 10 digits that are passed  as anchors. Why they are 10 in count, what these values represent and how I can obtain/calculate their values
3--- how to covert any tflite model with nncase to kmodel v3 (as I understanr with help of nncase 0.1)
4--- where to finf some normal documentation because tthat on sipeed webside is awful and not full
5--- what is doing operational feature map code from maixpy KPU documentation, why it is not working and how to fix it and make it to work. As model I am putting there 20calss.komedel from yolo demo
import sensor
import image
import lcd
import KPU as kpu
index=3  
lcd.init()
sensor.reset()
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QVGA)
sensor.run(1)
task=kpu.load(0x300000)
img=image.Image()
info=kpu.netinfo(task)
layer=info[index]
w=layer.wo()
h=layer.ho()
num=int(320*240/w/h)
list=[None]*num
x_step=int(320/w)
y_step=int(240/h)
img_lcd=image.Image()
while True:
    img=sensor.snapshot()
    fmap=kpu.forward(task,img,index)
    for i in range(0,num):
        list[i]=kpu.fmap(fmap,i)
    for i in range(0,num):
        list[i].stretch(64,255)
    for i in range(0,num):
        a=img_lcd.draw_image(list[i],((i%x_step)*w,(int(i/x_step))*h))
       lcd.display(img_lcd)
       kpu.fmap_free(fmap)

AIWintermuteAI, 2020-05-19:
@easy-and-simple  please use tags for mentioning people on GitHub. 
First of all, congratulations on beginning your journey to learning AI and coding. Your frustration is understandable - K210 is not a suitable chip for beginners. It is true, their documentation is lacking, but they are a small company, nothing like Espressif or Arduino. SO, I believe they're doing the best they can. Ranting and scolding WILL NOT help anybody.

Now going back to subject matter. You can execute script in serial monitor by clicking on Tools - Open Terminal in OpenMV IDE. Then to paste full script you need to press Ctrl-E once you get to REPL, paste your script and then press Ctrl+D to execute.

https://en.bbs.sipeed.com/t/topic/1790/15
set_output is explained here.
kpu.set_outputs(task, output_idx, w,h,ch)  0 is output_idx

Regrading the speed of kmodel v3 vs kmodel v3, the tests I did, didn't show significant difference in FPS. https://github.com/AIWintermuteAI/aXeleRate/issues/4

Finally, you are welcome to try the framework I wrote for training and converting the models, https://github.com/AIWintermuteAI/aXeleRate. If you encounter issues with it, post them on aXeleRate Github, but remember to be polite and explain the issue clearly.

AIWintermuteAI, 2020-05-19:
https://www.defmacro.org/2013/04/03/issue-etiquette.html @easy-and-simple  
I started with Machine Learning in 2016. It took me about 2 years of constant practice to get the hang of it and I don't consider myself an expert. So, you'll need a lot of patience and time to self-study.

1--- I don't know how to use microphone on K210, never used it
2--- YOLO v2 is not related in any way to K210, understanding how it works is your responsibility. Take a time to study https://github.com/pjreddie/darknet/issues/568 and other information you can find on Google
3--- there is no evidence that kmodel v3 is significantly faster than v4. Can you provide a reproducible example? like here https://github.com/AIWintermuteAI/aXeleRate/issues/4#issuecomment-626811776
4--- I have made a few videos and tutorials about K210, among other hardware. You can check hardware.ai channel on Youtube. documentation wise, I think their documentation is okay. it's not great, but in my experience it never is.
5--- If your encounter a new problem, please create a new issue and describe your problem there - DO NOT put every issue you have here, it is difficult for other people to follow up

Vinschni, 2020-05-22:
There is no encryption or blocking on that. All required to deploy your own models is opensource.

We are building firmware as documented and using tensorflow and k210 provided tools to run our own models. There are tutorials and examples on the internet.

Example: https://www.instructables.com/id/Transfer-Learning-With-Sipeed-MaiX-and-Arduino-IDE/

Issue can be closed.

easy-and-simple, 2020-05-22:
No cant be closed because only your models are running and models that are not made by you are not running. 
No documentation, no examples, all contains error and cant run, no info about firmware builds.
No example ho to use microphone. Speach recognition example is not working even with latest firmware 63
Source cant be build because your instructions are not correct
Give me example how to convert tflife to kmodel3 and dont send me to read nonsenses from 3-rd [arty sites
Also example code how to read model structure is not working

What is explanation that only models that you supply ready for use are working except that firmware supports only encrypted models?

easy-and-simple, 2020-05-22:
OK I finally succeed to run model that is not supplied form you.
If I do it at least one more time with other model I will close this thread, but you didn't helped me at all

closed 2020-05-23 by easy-and-simple

AIWintermuteAI, 2020-05-26:
https://github.com/sipeed/MaixPy/issues/242#issuecomment-633937461