sipeed/MaixPy-v1 #249: OV2640 cropping full resolution, understanding the registers

luismeruje, 2020-05-18:
I'm working with a sipeed maix go, and I'm trying to process UXGA pictures. The chip can't handle having more than VGA pictures in memory, so I want to only retrieve a "crop" of the full resolution at a time.
To do so, I'm trying to use the registers described in the datasheet: https://www.uctronics.com/download/cam_module/OV2640DS.pdf. The thing is I don't understand how the registers relate to each other, and I'm getting a lot of "sensor timeouts", "green images" and other things of the sort.
I've identified the following registers as being relevant: HSIZE, VSIZE, XOFFL, YOFFL, VHYX, TEST, ZMOW, ZMOH, ZMHH, SIZEL, HSIZE8, VSIZE8, COM1, COM7, HREFST, HREFEND, VSTRT, VEND and REG32.

Can someone help me understand how this registers work and how to use them to achieve the desired objective?

easy-and-simple, 2020-05-20:
For what purpose you need UXGA image and why you want to take it piece by piece??

luismeruje, 2020-05-20:
I want to capture distant objects with the best possible resolution.

easy-and-simple, 2020-05-20:
If you only need to make photo then you can set camera in jpeg mode. Compressed images need less memory. If image is not compressed it needs H*W*3 bytes.

luismeruje, 2020-05-20:
@easy-and-simple  thanks for your advice, but can I pass images through the kpu in jpeg format?

easy-and-simple, 2020-05-20:
Hahaha 
You said you only need to take part of picture with good resolution and didn't said you will process it. This is why I asked you why you want to do.
There is super simple solution for your problem but I dont want to say it here because guys are very toxic and refuse to help me and talk only useless things, but I can tell you want you need somewhere else

easy-and-simple, 2020-05-20:
By the way how you will process image in KPU if it is on pieces that you cant combine because of insufficient memory? Also KPU cant process images larger than 320x240

luismeruje, 2020-05-21:
I want to run a classification model on 224x224 sub-images within the main image. Those images are supposed to come from the full UXGA picture. I will have coordinates for each of those sub-images within the full resolution picture. 

I'm guessing I could do something like fetch the image in JPEG format and then crop it to one of the sub-images, iterating over a different sub-image on each frame, and convert the sub-image to RGB. However, that would probably be very inefficient and  I'm guessing since JPEG is compressed I would lose quality.

I think getting the sub-images I want directly from the sensor would be the cleanest way, and it wouldn't incur in quality loss, unless you know some other way?

easy-and-simple, 2020-05-21:
I see
I can open issue in some of your repos and to tell you how to do it. Is it ok for you?

luismeruje, 2020-05-21:
@easy-and-simple  Sure, I'll fork this repository

easy-and-simple, 2020-05-21:
but you have repos already I will open issue in one of them

luismeruje, 2020-05-21:
@easy-and-simple either way is ok, thanks for the help.

luismeruje, 2020-05-21:
I managed to find the correct register values. Turns out I had actually found them, but the image was just too dark, so I thought they were wrong. I set exposure manually and increased brightness to achieve a usable image.

closed 2020-05-21 by luismeruje