sipeed/MaixPy-v1 #250: Right sound of I2S is noisy, when I play *.wav.

galaxystar1yuji, 2020-05-21:
When I run "play_wav.py", then right sound is noisy. Left sound is no problem.
So I check the souce code of MaixPy, then I found the mistakes. I correct them,
I can hear a clear sound. Please correct like as follows.

File name : components/micropython/port/src/audio/wav/wav.c

(1)dst[i] = (src[i]<<16) | src[i];
       ->
    dst[i] = ((int32_t)src[i]<<16) | ((int32_t)src[i] & 0xFFFF);

(2)audio_buf[i] = ( MSB_audio << 16 ) | LSB_audio;
       ->
    audio_buf[i] = ( (int32_t)MSB_audio << 16 ) | ((int32_t)LSB_audio & 0xFFFF);

[wav.zip](https://github.com/sipeed/MaixPy/files/4662685/wav.zip)

junhuanchen, 2020-05-26:
Show me your python code?

galaxystar1yuji, 2020-05-26:
Python code is "play_wav.py" at https://github.com/sipeed/MaixPy_scripts , and I play stereo wav files.
Most important thing is that you need Maix board with stereo speaker, like as Maix Go or Maixduino with 2 speakers.
I connect PT8211, stereo amp and 2 speakers to Maix Bit.
Best regards.

junhuanchen, 2020-05-26:
This my test code , maybe you need to check the hardware.

```python
from Maix import GPIO, I2S, FFT
import image, lcd, math
from board import board_info
from fpioa_manager import fm
import audio

sample_rate = 38640
sample_points = 1024

fm.register(8,  fm.fpioa.GPIO0, force=True)
wifi_en=GPIO(GPIO.GPIO0,GPIO.OUT)
wifi_en.value(0)

fm.register(12,fm.fpioa.I2S0_IN_D0, force=True)
fm.register(13,fm.fpioa.I2S0_WS, force=True)    # 19 on Go Board and Bit(new version)
fm.register(14,fm.fpioa.I2S0_SCLK, force=True)  # 18 on Go Board and Bit(new version)

rx = I2S(I2S.DEVICE_0)
rx.channel_config(rx.CHANNEL_0, rx.RECEIVER, align_mode = I2S.STANDARD_MODE)
rx.set_sample_rate(sample_rate)

fm.register(0, fm.fpioa.GPIO1, force=True)
wifi_en=GPIO(GPIO.GPIO1, GPIO.OUT)
wifi_en.value(1)

# register i2s(i2s0) pin
fm.register(33,fm.fpioa.I2S2_OUT_D1, force=True)
fm.register(35,fm.fpioa.I2S2_SCLK, force=True)
fm.register(34,fm.fpioa.I2S2_WS, force=True)

# init i2s(i2s0)
wav_dev = I2S(I2S.DEVICE_2)

wav_dev.channel_config(wav_dev.CHANNEL_1, I2S.TRANSMITTER,resolution = I2S.RESOLUTION_16_BIT ,cycles = I2S.SCLK_CYCLES_32, align_mode = I2S.RIGHT_JUSTIFYING_MODE)
wav_dev.set_sample_rate(sample_rate)

import time

# loop to play audio
while True:
    wav_dev.play(rx.record(sample_points))
    print("running")
wav_dev.finish()

```

I directly record and forward the output to confirm the audio channel, instead of using wav output.

junhuanchen, 2020-05-26:
> When I run "play_wav.py", then right sound is noisy. Left sound is no problem.
> So I check the souce code of MaixPy, then I found the mistakes. I correct them,
> I can hear a clear sound. Please [...]

Play AVI, just like the information you provided, you can record wav and check it on the PC. The output PCM waveform data is divided into 8 and 16 bits for left and right channels.

galaxystar1yuji, 2020-05-27:
Thank you for your reply.  Your sample program is not for Maix Bit (with Mic). So I change
the pin numbers like as follows.
  Then I can hear the sound of mic from both left and right speakers. My system work well.

  When I play AVI, sound is no problem. That's because the source code of AVI sound is correct.
I already check them. Only "wav.c" have mistakes.
------------------------------------------------------
from Maix import GPIO, I2S, FFT
import image, lcd, math
from board import board_info
from fpioa_manager import fm
import audio

sample_rate = 38640
sample_points = 1024

fm.register(8,  fm.fpioa.GPIO0, force=True)
wifi_en=GPIO(GPIO.GPIO0,GPIO.OUT)
wifi_en.value(0)

fm.register(20,fm.fpioa.I2S0_IN_D0, force=True)
fm.register(19,fm.fpioa.I2S0_WS, force=True)    # 19 on Go Board and Bit(new version)
fm.register(18,fm.fpioa.I2S0_SCLK, force=True)  # 18 on Go Board and Bit(new version)

rx = I2S(I2S.DEVICE_0)
rx.channel_config(rx.CHANNEL_0, rx.RECEIVER, align_mode = I2S.STANDARD_MODE)
rx.set_sample_rate(sample_rate)

fm.register(0, fm.fpioa.GPIO1, force=True)
wifi_en=GPIO(GPIO.GPIO1, GPIO.OUT)
wifi_en.value(1)

# register i2s(i2s0) pin
fm.register(34,fm.fpioa.I2S2_OUT_D1, force=True)
fm.register(35,fm.fpioa.I2S2_SCLK, force=True)
fm.register(33,fm.fpioa.I2S2_WS, force=True)

# init i2s(i2s0)
wav_dev = I2S(I2S.DEVICE_2)

wav_dev.channel_config(wav_dev.CHANNEL_1, I2S.TRANSMITTER,resolution = I2S.RESOLUTION_16_BIT ,cycles = I2S.SCLK_CYCLES_32, align_mode = I2S.RIGHT_JUSTIFYING_MODE)
wav_dev.set_sample_rate(sample_rate)

import time

# loop to play audio
while True:
    wav_dev.play(rx.record(sample_points))
    print("running")
wav_dev.finish()

junhuanchen, 2020-05-27:
There are some defects in playing *.wav, and there are differences in the configuration of I2S. The import of WAV function based on the Python Code I provided may help you troubleshoot problems.

galaxystar1yuji, 2020-05-28:
No. You have a mistake.
You know that MSB_audio and LSB_audio are singed. So if MSB_audio=2 and LSB_audio=-1, then  ( MSB_audio << 16 ) | LSB_audio = (2 <<16) | (0xFFFFFFFF) = 0xFFFFFFFF. It's wrong. Correct value is 0x0002FFFF. Then left sound is 0xFFFF=-1 and right sound is 2.
So you have to change "(LSB_audio & 0xFFFF)"

junhuanchen, 2020-05-28:
oops，maybe you're right. where is it in wav.c?
I have some impressions about this.

galaxystar1yuji, 2020-05-28:
Please see the first comment. You have to change 2 lines.

(1) line 248
     dst[i] = (src[i]<<16) | src[i];
 and
(2)line 261
    audio_buf[i] = ( MSB_audio << 16 ) | LSB_audio;

case (1), when src[i]=-4, (src[i]<<16) | src[i] = (0xFFFFFFFC <<16) | 0xFFFFFFFC=0xFFFFFFFC. 
Correct result is 0xFFFCFFFC. So you need "(src[i] & 0xFFFF)".
  I check my system by kendryte SDK sample of play_pcm and sd_card, and can hear the clear sounds. But only play_wav of MaixPy is noisy. So I check the source code.
Best regards.

Neutree, 2020-08-06:
that's right, it's bug, thanks very much!

Neutree, 2020-08-06:
fix in https://github.com/sipeed/MaixPy/commit/4ca6d93b85cd50401ece9e8dee3ed61f5ce3945b

closed 2020-08-06 by Neutree