sipeed/MaixPy-v1 #251: ModuleNotFoundError: No module named 'serial'

easy-and-simple, 2020-05-22:
Why I get error when Try to compile maixpy from source?

maixpy
Traceback (most recent call last):
  File "project.py", line 31, in <module>
    exec(f.read())
  File "<string>", line 37, in <module>
  File "/home/user1/maixpy/MaixPy/tools/flash/flash.py", line 12, in <module>
    import serial.tools.miniterm
ModuleNotFoundError: No module named 'serial'

easy-and-simple, 2020-05-22:
How to build documentation is bad and misleading and cant be follower step by step because some steal are missed, and should improvise

closed 2020-05-22 by easy-and-simple

GuilhermeAgostin, 2020-06-30:
I have the same problem, can you help me solve it?

[image]

easy-and-simple, 2020-07-01:
sudo pip3 install -r requirements.txt

or pip install pyserial==3.4