sipeed/MaixPy-v1 #253: Memory problem with maixpy 5 and loaded 3 face models

minyatos, 2020-05-25:
I have RAM problem with new versiyon of maixpy_v0.5.0_63_g2d307ae.bin & FD_,FE_ and KP_ face models. 
They can not work together on Maix Go. But I also need both face recognation and QRcode.
If I load a much older versiyon maixpy I can load all 3 models but then  find_qrcodes ,find_blob functions won't work.

easy-and-simple, 2020-05-28:
did you tried minimum firmware? If yes then Init and deinit them one by one so only 1 of the to be active at same time

minyatos, 2020-05-29:
Yes, i will try to deinit them as you said, i will inform you about process thank you

Windows 10 için Posta<https://go.microsoft.com/fwlink/?LinkId=550986> ile gönderildi

Kimden: easy-and-simple<mailto:notifications@github.com>
Gönderilme: 28 Mayıs 2020 Perşembe 09:29
Kime: sipeed/MaixPy<mailto:MaixPy@noreply.github.com>
Bilgi: minyatos<mailto:gozdey@msn.com>; Author<mailto:author@noreply.github.com>
Konu: Re: [sipeed/MaixPy] Memory problem with maixpy 5 and loaded 3 face models (#253)

did you tried minimum firmware? If yes then Init and deinit them one by one so only 1 of the to be active at same time

minyatos, 2020-05-30:
[image]
Hi again, i tried kpu.deinit(task) function and it didn't work somehow cause I'm using your 3 model face recognition and getting images from camera. 
Then I tried kpu_loadflash function, it didn't work either when I convert the 3 models with convert.py function you provided. Since I couldn't see inside the models, I can't say what is wrong.  
Then I tried to compile my own maixpy with my own requirements, and include qrcode functionality, but it din't work with face model project. I though maybe I missed to include something and add lib_sipeed_kpu option and it gets error at the following code block I attached.
So can you please help me with;
1-  how can I compile maixpy_minimum version with qrcode functionality and also support the 3 face model example? What is the dependencies, that my version won't work at face sample?
2- If it needs compile of lib_sipeed_kpu, can you please correct micropython requirements code for me to compile. (lib_sipeed_kpu.a ?)  
3- why I can not convert FD,FE and KP face models for load from flash functionality and couldn't run them? Is it about models, if yes what is the correct face models for me to use flash functionality? 
Thank you :)

easy-and-simple, 2020-05-31:
I have no idea what are you doing and what is causing your problem but can advice you what you could try if this is low memory problem
Try this, save script to board as boot.py disconnect from ide and reboot board via reset key
Also when you compile your firmware version most important is to uncheck double buffering option

minyatos, 2020-06-01:
Can you provide working "convert.py" function for the below face models?
I want to use _kpu.load_flash(0x400000, 0, 0,50000000)_

[image]

closed 2020-06-02 by minyatos