sipeed/MaixPy-v1 #254: Getting frame from video

AndreiUngureanu101, 2020-05-26:
Hello,
Is there any way of getting each frame from the video and running this stream through the detection models, for example?

easy-and-simple, 2020-06-02:
import os, sys, time, image, lcd

f = open("badapple.avi", "rb")
index = 0
tmp = f.read(4)
while tmp !=b'movi':
    index += 1
    a = f.seek(index,0)
    tmp = f.read(4)
index += 4
movi_location = index
f.seek(0,2)
f_size = f.tell()
index = f_size
while tmp != b'idx1':
    index -= 1
    a = f.seek(index, 0)
    tmp = f.read(4)
count = 0
lcd.init()
tim = time.ticks_ms()
while tmp:
    while tmp != b'00dc':
        index += 1
        a = f.seek(index, 0)
        tmp = f.read(4)
    index += 8       #dwflag +4bytes from dwflag--->dwoffset=frame_loaction
    a = f.seek(index,0)
    tmp = f.read(4)
    frame_location = movi_location + int.from_bytes(tmp, "little")
    a = f.seek(frame_location,0)
    tmp = f.read(4)
    frame_size = int.from_bytes(tmp, 'little')
    frame_location += 4 #//only data length removed
    a = f.seek(frame_location, 0)
    img = image.Image(f, copy_to_fb=1)
    a = img.draw_string(2,2,"%.2ffps" %(1000/(time.ticks_ms()-tim)), color=(255,0,0), scale=2)
    lcd.display(img)
    a = f.seek(index, 0)
    count += 1
    tim = time.ticks_ms()

f.close()

AndreiUngureanu101, 2020-06-02:
@easy-and-simple Thanks but this is not formatted so it's hard to try to understand what's going on.

easy-and-simple, 2020-06-02:
Now it is formated

closed 2020-07-31 by Neutree

diazGT94, 2021-01-19:
Hello, I'm trying to use this script with my own video. However, it's stay tucked in this  while tmp != b'00dc' since the tmp never gets a value equal to 00dc

diazGT94, 2021-04-21:
How can this script be adapted to Maixpy3? I've trying to do it but so far I haven't been able to display the image using the PIL library