sipeed/MaixPy-v1 #257: lcd color format is complex and dumb, because doesn't match image color format

easy-and-simple, 2020-05-30:
Why image color is tuple and lcd color is int16?!
This is really dumb!!!
In case to draw something in lcd and image with same color we need to write whole program. WTF?!
I think lcd colors should be unified with image colors and to be set in code as 3 bytes tuple too

Neutree, 2020-07-31:
use image to draw anything is recommend, lcd just for some special use, but we can add RGB888 support along with RGB565

easy-and-simple, 2020-07-31:
I am not talking for image format, but for color format difference when working with image and LCD
For example if I want to draw rectangle on image and write text on LCD as in your yolo example script, and I want to do it with different color for different classes I can't use same color value because for image RED color is tuple (255,0,0) while for LCD RED color is integer 0x00F8. 
I think that both should be in form of tuple (255,0,0) because it is universal and easy to set desired color

Neutree, 2020-08-03:
that's what I said, arg is RGB888(e.g. (255, 0, 0)) or RGB565(e.g. 0xffff)