sipeed/MaixPy-v1 #258: Image font unacceptable small and useless

easy-and-simple, 2020-05-30:
Image font is unacceptable small and useless
Fake 8x10 pixels
Real 4x6 pixels WTF?! Realy useless. Nothing can't be written in 1x scale and 2x scale looks really ugly. Words describing look of higher scales are not suitable for kids
It should be at least 2x that means atleast 8x12 real pixels and should be unified in size with size of LCD font

Neutree, 2020-07-31:
we have update the font in develop branch @junhuanchen

junhuanchen, 2020-07-31:
Now

The Image.font change 8 * 12

[image]

[image]

Maybe you can try the latest.

easy-and-simple, 2020-07-31:
Is it already working?
If yes why then these changes are only in development branch?
When changes will be available in master branch?

junhuanchen, 2020-08-01:
> Is it already working?
> If yes why then these changes are only in development branch?
> When changes will be available in master branch?

Need to fine tune the font effects.

easy-and-simple, 2020-08-01:
OK thank you for this information!
Will wait update of master branch

junhuanchen, 2020-08-20:
> OK thank you for this information!
> Will wait update of master branch

now Merge branch 'master' of https://github.com/sipeed/MaixPy into develop

https://github.com/sipeed/MaixPy/commit/8158d48704c0225790bc03dcc1affc259b364736

If you have other problems, you can reopen it.

closed 2020-08-20 by junhuanchen