sipeed/MaixPy-v1 #259: openMV

halilkalayli, 2020-05-30:
maixpy’s openmv implementation is not up-to-date. there are functions that is not implemented. when will they updated?

easy-and-simple, 2020-05-31:
No such thing like openmv
There are openmv ide and openmv cam and some openmv api that is specific to openmv cam boards.
Maybe you will tell us what you call openmv and what functions are not up to date and how you found that if maixpy documentation is so bad and incomplete?

halilkalayli, 2020-05-31:
Sorry my mistake. I wanted to say openMV library. 
for example:
- image.crop() function returns AttributeError: 'Image' object has no attribute 'crop'
- blob.corners() function returns AttributeError: 'blob' object has no attribute 'corners'
- blob.min_corners() function returns AttributeError: 'blob' object has no attribute 'min_corners'
- image.rotation_corr() has no option for "fov" and "corners"

my setup:
- sipeed m1 dock suit
- MaixPy IDE 0.2.5
- firmware: maixpy_v0.5.0_88

easy-and-simple, 2020-05-31:
You are right that there are a lot of things that should be fixed, implemented and complete, but I think your problem is in bad documentation.
I got similar errors when tried to use functions directly on objects, but it is working when you assign object to variable and then use desired function. 
I don't know is there is crop function for image because in examples I have seen cut function, but if there is function crop you should not call it directly as in your example - image.crop, but should use something like var1=image.image and then var1.crop. I hop you got the idea

halilkalayli, 2020-05-31:
Of course there are some documentation problems. But what i wanted to point is openMV has updates (as seen from attachments) and sipeed did not implement them. 

[image]
[image]

junhuanchen, 2020-07-01:
Mark, these interfaces are really not implemented.

FPSychotic, 2020-07-15:
Please add the thermal camera drivers that openmv support