sipeed/MaixPy-v1 #263: Not detecting anything with custom kmodel

nicolaivalenti, 2020-06-05:
I'm using this example (link below) with the Pascal Voc dataset, I have not changed anything (I trained for 100 epochs as standard in the notebook). The architecture is Mobilenet 1_0.

https://colab.research.google.com/github/AIWintermuteAI/aXeleRate/blob/master/resources/aXeleRate_pascal20_detector.ipynb

[image]

I'm able to run with the code you can see in the previous message, but it does not detect anything, also if I open the serial terminal while the code is running it continuously prints numbers from 7 to 9 with 6 decimals. **The model seems to work when I test it using the last cell of the notebook in Coolab, it gives me most of the test images with the correct detections.**
It's probably something that I'm forgetting because I'm a newbie.

easy-and-simple, 2020-06-07:
In order model to detect objects, there must be object. On you screenshot there is no any objects what you expect to detest?

nicolaivalenti, 2020-06-07:
> In order model to detect objects, there must be objects. On you screenshot there is no any objects. What you expect to detect on blank image?

I'm not that dumb, I posted the screenshot just to show the code.
I tried with all the different classes using images from google

easy-and-simple, 2020-06-07:
You call fps numbers and supplied screenshot without objects. Obviously you have no any idea what are you doing. In this case ofcourse it will not work
Code is of yolo 20 objects demo, and it is not modified  so screenshot showing it is not giving any useful info
You should give some useful info if you want someone to help you with this

easy-and-simple, 2020-06-07:
If you dont understand anything yet, at first try example scripts that are working like yolo2 20 classes demo to see how things works and then try to make your custom model
yolo2 20 classes example is doing the same as what you are trying to do
Get demo scripts from here https://github.com/sipeed/MaixPy_scripts
In machine vision folder open yolov2_20class.py
Inside you will find link from where to download working kmodel for this demo

closed 2020-07-04 by myuanz