sipeed/MaixPy-v1 #267: OSError: run error: 13

niketmalusare, 2020-06-20:
I am trying custom trained .kmodel on maix_bit
i always get error OSError: run error: 13 at line a = kpu.init_yolo2(task, 0.3, 0.3, 5, anchor)
i tried loading .kmodel from sd card but still get same error
my code is as follows 

import sensor,image,lcd
import KPU as kpu

lcd.init()
sensor.reset()
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QVGA)
sensor.set_windowing((224, 224))
sensor.set_vflip(1)
sensor.run(1)
classes = ["bearing_not_ok","bearing_ok"]
task = kpu.load(0x300000)
#task = kpu.load('/sd/m7_5.kmodel')
anchor = (0.57273, 0.677385, 1.87446, 2.06253, 3.33843, 5.47434, 7.88282, 3.52778, 9.77052, 9.16828)
a = kpu.init_yolo2(task, 0.3, 0.3, 5, anchor)
while(True):
    img = sensor.snapshot()
    #.rotation_corr(z_rotation=90.0)
    #a = img.pix_to_ai()
    code = kpu.run_yolo2(task, img)
    if code:
        for i in code:
            a=img.draw_rectangle(i.rect(),color = (0, 255, 0))
            a = img.draw_string(i.x(),i.y(), classes[i.classid()], color=(255,0,0), scale=3)
        a = lcd.display(img)
    else:
        a = lcd.display(img)
a = kpu.deinit(task)

easy-and-simple, 2020-06-23:
If this is kmodel4 you should execute kpu.set_outputs before kpu.init_yolo2

myuanz, 2020-07-01:
Is the problem solved now?

Neutree, 2020-07-31:
`kpu.init_yolo2` just init yolo parameters, no kpu related opration, and no OS error raised, maybe not wrong in this func?

closed 2020-07-31 by Neutree

beckmx, 2023-02-15:
@Neutree hello, I took an example from the scripts section that reads like this:
```
clock = time.clock()
classes = ['aeroplane', 'bicycle', 'bird', 'boat', 'bottle', 'bus', 'car', 'cat', 'chair', 'cow', 'diningtable', 'dog', 'horse', 'motorbike', 'person', 'pottedplant', 'sheep', 'sofa', 'train', 'tvmonitor']
task = kpu.load(0x500000) #change to "/sd/name_of_the_model_file.kmodel" if loading from SD card
anchor = (1.08, 1.19, 3.42, 4.41, 6.63, 11.38, 9.42, 5.11, 16.62, 10.52)
a = kpu.init_yolo2(task, 0.5, 0.3, 5, anchor)
print("is yolo loaded:")
print(a)
while(True):
    clock.tick()
    img = sensor.snapshot()
    code = kpu.run_yolo2(task, img)####<<<<<----break here
```

But it brakes in the line selected in there, according to the documentation it says, that is wrong but it doesnt say how to correct it or what should be do instead here: https://wiki.sipeed.com/soft/maixpy/en/api_reference/Maix/kpu.html#run_yolo2 do you have an updated doc or example for that?