sipeed/MaixPy-v1 #268: kpu load error -3

pritamghanghas, 2020-06-28:
While we try to load model, we get kpu error. From The MaixPy sources, I couddn't find what -3 means, there all enums are +ve.

task = kpu.load('/sd/model.kmodel')
ValueError: [MAIXPY]kpu: load error -3 from this line

model can be accessed at this location https://drive.google.com/file/d/1qorFP6I7mo5T_cVk3XzfeY2XULG9LGah/view?usp=sharing

myuanz, 2020-07-01:
Plz tell us the Maix firmware version.

pritamghanghas, 2020-07-01:
It gave -3 error on maixpy_v0.5.0_95_g05c4e70_with_lvgl.bin. Later I loaded the firmware attached in this blog post https://en.bbs.sipeed.com/t/topic/1790, I think it was first firmware that supports v4. In this firmware the model loaded bug the call to kpu.forward() hangs up and nothing happens unless I rest the board.

myuanz, 2020-07-01:
Thanks for your feedback, this issue has been added to our todo list. 😅

pritamghanghas, 2020-07-01:
Let me know if tflite model is also need for any verification, I can upload that as well.

myuanz, 2020-07-04:
No, you can convert any model supported by nncase.

Neutree, 2020-07-30:
@pritamghanghas Now you should customsize your firmware according the build.md, or generate a firmware by online compile tool(https://www.maixhub.com/compile.html), make sure kmodel v4 support is checked

I tried your kmodel, it's ok
[image]

If there remains error, reopen this issue

closed 2020-07-30 by Neutree