sipeed/MaixPy-v1 #269: FFT not working

valeumvale, 2020-06-30:
I was on using the fft demo on v0.3.2, but then I wanted to try the speechRecognition demo and it didn't work because there is no  speechRecognition file on 'modules', so i updated the firmware to v0.5.0_95_g05c4e70, which still didn't work and now the FFT demo that I was using on a script always returns 0 on the new firmware, and on the old the "force=true" used on GPIOs comands return as an error.
Can someone help me?

valeumvale, 2020-06-30:
Im using maispy dock and the script works on v0.3.2, but after I send it to the board as boot.py it doesn't run out of Maix IDE

myuanz, 2020-07-01:
Have a try about [0.5.0](https://dl.sipeed.com/MAIX/MaixPy/release/master)? May be a solved problem

valeumvale, 2020-07-01:
Problem solved.
I was using the latest 0.5.0, but I had comented "lcd.display(img)" because I didn't need the display but it looks like I can´t, I'll be changing be code so it doesn´t need a display.

closed 2020-07-01 by valeumvale