sipeed/MaixPy-v1 #270: Error building Image: 'pxStackTop' not defined.

robert-hh, 2020-07-01:
Whenbuilding the image as I used to do, I run into an error:
```
/home/robert/Downloads/MicroPython/MaixPy/components/micropython/port/src/mpthreadport.c: In function 'mp_thread_gc_others':
/home/robert/Downloads/MicroPython/MaixPy/components/micropython/port/src/mpthreadport.c:88:45: error: 'TaskStatus_t' {aka 'struct xTASK_STATUS'} has no member named 'pxStackTop'; did you mean 'pxStackBase'?
         gc_collect_root((void**)task_status.pxStackTop, ( (uint64_t)th->stack + (th->stack_len*sizeof(StackType_t)+1024) - (uint64_t)task_status.pxStackTop)/sizeof(void*));
                                             ^~~~~~~~~~
```

robert-hh, 2020-07-01:
Forgot submodule update - not the first time.

junhuanchen, 2020-07-01:
you can pxStackTop change to pxStackBase.

or change components/kendryte_sdk/kendryte-standalone-sdk branch to Maixpy-dev

sorry~

closed 2020-07-04 by myuanz