sipeed/MaixPy-v1 #273: KPU: Is there any way to run a CNN with more than 3 input channels?

foxis, 2020-07-11:
I'm currently developing a SLAM algorithm that I would like to run on K210 and for this, I need to input 2-3-more images to the network at the same time (stacked by channels).
I've browsed through the code and got as far as `lib_sipeed.a`. I can see it could be possible to modify `py_image` to be able to stack images(lots of workarounds) or modify `maixpy_kpu` to accept several images(fewer workarounds), however, I'm not sure how RGB565 is being split into individual bytes inside the closed source library which is a blocker to my investigations...

Any suggestions?

Neutree, 2020-07-30:
change the second arg of sipeed_kpu_model_run() may ok?

Neutree, 2020-07-30:
default is img->pix_ai, but you change to another