sipeed/MaixPy-v1 #274: i2c not working properly

pritamghanghas, 2020-07-12:
My model loads only in this firmware linked in  blog post https://en.bbs.sipeed.com/t/topic/1790.
All other firmwares it either crash, returns error or simply hang up either in load or run. 
In the working firmware i2c doesn't recognize mlx90614, it is not a micropython problem because I tried it on esp8266 and it works like a charm. 
It is not wiring problem either as I can run a i2c lcd with same pins.

Neutree, 2020-07-30:
What "i2c doesn't recognize mlx90614" means?, `i2c.scan()` return empty?
maybe you can directly access to device, share your code

pritamghanghas, 2020-08-02:
scan returns empty but I am able to access it directly because the address is already known.

junhuanchen, 2020-09-14:
Now support software i2c

can you try I2C3+?

```python

    from machine import I2C
    i2c = I2C(I2C.I2C3, freq=100*1000, scl=24, sda=25)
```

This is a hardware design flaw.

junhuanchen, 2020-09-19:
If there is still an I2C device that cannot be found, please reply to the device model and reopen this issue, thanks.

closed 2020-09-19 by junhuanchen