sipeed/MaixPy-v1 #275: execption while model loading

pritamghanghas, 2020-07-12:
maixpy_v0.5.0_98_g7ec09ea_with_lvgl leads to exception while loading my model. 

> init i2c2                                                                                                                                                                  
> [MAIXPY]: find ov2640                                                                                                                                                      
> core dump: fault load                                                                                                                                                      
> Cause 0x0000000000000005, EPC 0x000000008000116e                                                                                                                           
> reg[00](zero ) = 0x0000000000000000, reg[01](ra   ) = 0x000000008000023e                                                                                                   
> reg[02](sp   ) = 0x00000000803206e8, reg[03](gp   ) = 0x0000000080232120                                                                                                   
> reg[04](tp   ) = 0x0000000000000000, reg[05](t0   ) = 0x0000001000000000
> reg[06](t1   ) = 0x000000000003fe12, reg[07](t2   ) = 0x000000000000002b
> reg[08](s0/fp) = 0x000000008011f8a4, reg[09](s1   ) = 0xbe2177b81307656b
> reg[10](a0   ) = 0x0000000000000000, reg[11](a1   ) = 0xbe2177b813076172
> reg[12](a2   ) = 0x0000000080320908, reg[13](a3   ) = 0x0000000080320a08
> reg[14](a4   ) = 0x0000000000000008, reg[15](a5   ) = 0xbe2177b813076172
> reg[16](a6   ) = 0x000000000000000b, reg[17](a7   ) = 0x0000000000000000
> reg[18](s2   ) = 0x0000000080309760, reg[19](s3   ) = 0x000000008042b2e0
> reg[20](s4   ) = 0x0000000000000000, reg[21](s5   ) = 0x000000008032e4e8
> reg[22](s6   ) = 0x000000008032ec20, reg[23](s7   ) = 0x000000008042b300
> reg[24](s8   ) = 0x000000008032e320, reg[25](s9   ) = 0x000000008022d398
> reg[26](s10  ) = 0x0000000000000001, reg[27](s11  ) = 0x0000000000000002
> reg[28](t3   ) = 0x0000000000000001, reg[29](t4   ) = 0x0000000000000000
> reg[30](t5   ) = 0x000000000003fe12, reg[31](t6   ) = 0x0000000000000003
> freg[00](ft0 ) = 0x0000000000000000(), freg[00]() = 0x000000008012cfa0()
> freg[02](ft2 ) = 0x0000000000000000(), freg[00]() = 0x000000008012cfb0()
> freg[04](ft4 ) = 0x0000000000000000(), freg[00]() = 0x000000008012cfc0()
> freg[06](ft6 ) = 0x0000000000000000(), freg[00]() = 0x000000008012cfd0()
> freg[08](fs0 ) = 0x0000000000000000(), freg[00]() = 0x000000008012cff8()
> freg[10](fa0 ) = 0x00000000408aaaab(), freg[-1610612736]() = 0x000000008012d028()
> freg[12](fa2 ) = 0x0000000041200000(), freg[00]() = 0x000000008012d050()
> freg[14](fa4 ) = 0x000000004bb71b00(), freg[-1073741824]() = 0x000000008012d060()
> freg[16](fa6 ) = 0x0000000000000000(), freg[00]() = 0x000000008012d070()
> freg[18](fs2 ) = 0x0000000000000000(), freg[00]() = 0x000000008012d098()
> freg[20](fs4 ) = 0x0000000000000000(), freg[00]() = 0x000000008012d0a8()
> freg[22](fs6 ) = 0x0000000000000000(), freg[00]() = 0x000000008012d0b8()
> freg[24](fs8 ) = 0x0000000000000000(), freg[00]() = 0x000000008012d0c8()
> freg[26](fs10) = 0x0000000000000000(), freg[00]() = 0x000000008012d0d8()
> freg[28](ft8 ) = 0x0000000000000000(), freg[00]() = 0x000000008012d100()
> freg[30](ft10) = 0xffffffff00000000(), freg[00]() = 0x000000008012d110()
> W (15955242709) SYSCALL: sys_exit called by core 0 with 0x539

the same model works with another firmware as reported in this issues https://github.com/sipeed/MaixPy/issues/268
but there i2c doesn't work properly.

Neutree, 2020-07-30:
maybe you used the firmware not support kmodel V4, compile firmware you need according I saied in #268, and try again

Neutree, 2020-08-10:
I tried with self compiled firmware with V4 and lvgl support firmware, it's ok when load your model

closed 2020-08-10 by Neutree