sipeed/MaixPy-v1 #279: Broken config should be reset

Vinschni, 2020-07-19:
I get:
```
read config fail

[MAIXPY]Pll0:freq:832000000
[MAIXPY]Pll1:freq:398666666
[MAIXPY]Pll2:freq:45066666
[MAIXPY]cpu:freq:416000000
[MAIXPY]kpu:freq:24916666
[MAIXPY]Flash:0xef:0x17
open second core...
```

I assume if config is broken it is not reqritten correctly and leads to invalid state.

junhuanchen, 2020-07-31:
https://github.com/sipeed/MaixPy/blob/49cb9b05142dbe34ae8fc0df8c5316b9e543d4bd/components/micropython/port/src/maixpy_main.c#L300-L325

SPIFFS_read & save_config_to_spiffs fail will try SPIFFS_format in develop.

If you have tested?

Vinschni, 2020-08-04:
I built 49cb9b0.
Checked that your fix is there:
https://github.com/sipeed/MaixPy/blob/49cb9b05142dbe34ae8fc0df8c5316b9e543d4bd/components/micropython/port/src/maixpy_main.c#L323

And got following:
```maixpy read config fail

[MAIXPY]Pll0:freq:832000000
[MAIXPY]Pll1:freq:398666666
[MAIXPY]Pll2:freq:45066666
[MAIXPY]cpu:freq:416000000
[MAIXPY]kpu:freq:24916666
[MAIXPY]Flash:0xef:0x17
open second core...
gc heap=0x8020ca60-0x8020ca60(0)
nlr_jump_fail```

Vinschni, 2020-08-04:
Would dumping my whole flash to an image to allow you reproduction on your MaixBit help?
If wished how can i dump whole flash simple?

junhuanchen, 2020-08-04:
mod_os_flash_format should be used.

I forget SPIFFS_unmount(&spiffs->fs); and SPIFFS_mount.

so try this function.

`MP_DEFINE_CONST_FUN_OBJ_0(mod_os_flash_format_obj, mod_os_flash_format);`

https://github.com/sipeed/MaixPy/blob/49cb9b05142dbe34ae8fc0df8c5316b9e543d4bd/components/micropython/port/src/standard_lib/uos/moduos.c#L148

junhuanchen, 2020-08-04:
try fix this. https://github.com/sipeed/MaixPy/commit/d0eeeb0559662a0549e16f394f6f6f9371f0ef3c

junhuanchen, 2020-08-04:
> Would dumping my whole flash to an image to allow you reproduction on your MaixBit help?
> If wished how can i dump whole flash simple?

you can use tool/spiffs/gen_spiffs_image.py and pack your image to flash.

Vinschni, 2020-08-04:
https://github.com/sipeed/MaixPy/commit/d0eeeb0559662a0549e16f394f6f6f9371f0ef3c#commitcomment-41164322

junhuanchen, 2020-08-05:
I decided that I should be able to use the default configuration to get into the system, even if any of the following problems occurred.

1. Flash damage of hardware chip

2. Freq.conf configuration file error, such as file content error

3. Spiffs Flash system error, such as file index error

The solutions are as follows:

1. If the chip is damaged, the hardware cannot be used.

2. Can be modified by the user after entering maixpy, such as' os.remove('freq.conf') '

3. It is up to the user to format the SPIFFS system, such as' os.flash_format() '

Maixpy can enter the REPL interface no matter what happens.

junhuanchen, 2020-08-05:
Fix this https://github.com/sipeed/MaixPy/commit/66bb1bab41bef26ef1c894e3a157f84b03352e76

junhuanchen, 2020-08-20:
If you have other problems, you can reopen it

closed 2020-08-20 by junhuanchen